FreeCAD/FreeCAD #23375: Angle measurement shows wrong value or angle

worldpeace-germany, 2025-08-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When measuring an angle it seems that the angle is either shown in the wrong position or has a wrong value

```
         /
--------/--------

```
In above sketch the left angle is lets say 100° and the right angle 80°. FreeCAD would show for the left angle 80°.

[image]

I could upload the sample file from the screenshot but not sure how to upload this file here. The normal add files tells me it is not supported file type.

### Full version info

```shell
OS: openSUSE Leap 15.6 (KDE/plasma5/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-08-26:
@worldpeace-germany can you test 1.1 development build ? https://github.com/FreeCAD/FreeCAD/releases and report back ?

worldpeace-germany, 2025-08-26:
Hello,

yes same issue on

OS: openSUSE Leap 15.6 (KDE/plasma5/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/20 00:00:55
Build type: Release
Branch: main
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5

Attached the sample file, I have just added the .zip extension I have not zipped the file. So just remove the .zip ending don't unzip it.

Thanks.

[Angle_Issue.FCStd.zip](https://github.com/user-attachments/files/21988585/Angle_Issue.FCStd.zip)

closed 2025-08-27 by luzpaz