FreeCAD/FreeCAD #23401: FreeCAD crashes in Sketcher

pmjdebruijn, 2025-08-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have the following issue with FreeCAD 1.1.0dev crashing in the sketcher, which does not happen with 1.0.2:

```
freecad: ./src/nodes/SoGroup.cpp:293: virtual SoNode* SoGroup::getChild(int) const: Assertion `(index >= 0) && (index < this->getNumChildren())' failed.
Aborted

freecad: ./src/elements/SoCoordinateElement.cpp:178: const SbVec3f& SoCoordinateElement::get3(int) const: Assertion `index >= 0 && index < this->numCoords' failed.
Aborted
```

If I'm not mistaken these are COIN assertions, right?

This is on Debian 13 (trixie), with libcoin80t64:amd64 version 4.0.3+ds-2

### Full version info

```shell
1.1.0dev from git checkout weekly 2025.08.27+6~g5af7ad5df4
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pmjdebruijn, 2025-08-27:
A sample to reproduce:

[CoinCrash.FCStd.zip](https://github.com/user-attachments/files/22010019/CoinCrash.FCStd.zip)

Just doubleclick on the Sketch in the object tree

Pendelf, 2025-08-27:
I can't reproduce it

https://github.com/user-attachments/assets/39a7d14d-dab2-4bc6-a516-dfa0482e943b

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.43071 (Git)
Build date: 2025/08/25 18:08:30
Build type: Release
Branch: main
Hash: d85613b498881c47485a7d930807673203086756
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC 1.9.1
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * fasteners 0.5.41
  * manifest.json
  * PieMenu 1.12.0
  * sheetmetal 0.7.24
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.2

pmjdebruijn, 2025-08-27:
Here's a backtrace, not sure if it's complete enough:

[freecad-coin-assert-backtrace.txt](https://github.com/user-attachments/files/22012211/freecad-coin-assert-backtrace.txt)

pmjdebruijn, 2025-08-28:
I've also just verified this isn't tied to any particular setting, with ~/.config/FreeCAD removed it still occurs...

matthiasdanner, 2025-08-29:
@pmjdebruijn could you try to record a screencast when it crashes?

luzpaz, 2025-08-29:
@pmjdebruijn can you please provide you full [About](https://wiki.freecad.org/About) info. TYI

ulmerkott, 2025-08-30:
I've managed to make a sketch which I think reproduces this 100% with FreeCAD 1.1.0, Libs: 1.1.0devR43087 (Git).

1. Open the first pocket sketch 
2. Try to select this tangent constraint

[image]

[mini_dremel_gridfinity_case.zip](https://github.com/user-attachments/files/22056716/mini_dremel_gridfinity_case.zip)

maxwxyz, 2025-08-30:
can you reproduce the crash in the stable 1.0 version?

maxwxyz, 2025-08-30:
I cannot reproduce, I can select it without problems.

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * AddonManager 2025.8.16
  * Cables 0.2.1
  * CurvedShapes 1.0.13
  * Defeaturing 1.2.6
  * dodo 1.0.1
  * freecad.gears 1.3.0
  * Gridfinity 0.11.9
  * manifest.json
  * Manipulator 1.6.0
  * OpticsWorkbench 1.2.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.24
  * Silk 0.2.0
  * Telemetry 1.0.2

```

pmjdebruijn, 2025-08-30:
> [@pmjdebruijn](https://github.com/pmjdebruijn) can you please provide you full [About](https://wiki.freecad.org/About) info. TYI

I can still reproduce the issue with current git main: 2025.08.27+45~g3b76d77ed8

```
OS: Debian GNU/Linux 13 (trixie) (XFCE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.
Build date: 2025/08/30 13:39:07
Build type: Release
Python 3.13.5, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: C/Default (C)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/unset/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/93.2987/1
```

matthiasdanner, 2025-08-30:
@pmjdebruijn

The crash from your backtrace is happening, when you are selecting a constraint and has only one icon (like the tangent constraint).
Then the line 2229 in EditModeConstraintCoinManager crashes (only with assertions enabled).
If the assertions are disabled you simply get a null value and this is handled by the if.

```
 SoNode* secondIconNode =
     sep->getChild(static_cast<int>(ConstraintNodePosition::SecondIconIndex)); // if there is no second Child this will be null or with assertions it crashes, but it is handled in the next line
 if (tail == secondIconNode) {
```

So,  I don't get why you would want to compile with assertions in the first place.

pmjdebruijn, 2025-08-30:
@matthiasdanner is there any you could give me a patch, so I can test that here locally?

matthiasdanner, 2025-08-30:
I dont think the assertions have to be handled at all costs because that has implications on its own Like performance or it could lead to more complicated Code. For me assertions Are only a Tool for developing. 

If You want to test it just add the Code snippet from my last comment on the said line.

ulmerkott, 2025-08-30:
Managed to reproduce it with [mini_dremel_gridfinity_case.zip](https://github.com/user-attachments/files/22056716/mini_dremel_gridfinity_case.zip) in a debug build in a Ubuntu container with default settings. So here's a bit more info.

OS: Ubuntu 25.04 ()
Architecture: x86_64
Version: 1.1.0dev.Unknown
Build date: Unknown
Build type: Unknown
Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: C/Default (C)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/110.786/1

Full backtrace:
[full_backtrace_mini_dremel_gridfinity_case.txt](https://github.com/user-attachments/files/22058455/full_backtrace_mini_dremel_gridfinity_case.txt)

Just ask if you want me to test anything further.

matthiasdanner, 2025-08-30:
Ok if its default on linux...

I'll push the Change later

pmjdebruijn, 2025-08-30:
@matthiasdanner if you push this fix already, then I'll do a rebuild included the one fix, at least then if I hit another crash, I know the stackstace I get, will be the other issue, not the one that we already know about...

matthiasdanner, 2025-08-30:
I created a pullrequest for the first crash.

I couldn't reproduce the get3 crash, since I don't think you can compile the libraries with the debug scheme on windows.

Could you describe what you do in order for this crash to occur?

pmjdebruijn, 2025-09-15:
@matthiasdanner with at least the one fix, the FreeCAD sketcher seems much more stable, I'll continue to try to reproduce...

However, is the maybe some debug code I could patch in locally to zero in one the issue, if I manage to reproduce eventually?

ulmerkott, 2025-10-25:
> [@ulmerkott](https://github.com/ulmerkott) did you have any crashes with recent git checkouts?
> 
> Because I haven't had any crashes in the Sketcher for a while now, so as far as I'm concerned the issue can be closed unless [@ulmerkott](https://github.com/ulmerkott) is still seeing crashes?

I've not experienced any crashes in the sketcher either after the fix. So I'm also fine with the issue being closed.

closed 2025-10-25 by maxwxyz