FreeCAD/FreeCAD #23402: Core: valid v1.0.2 file opens with errors in v1.1 (TNP?)

Roy-043, 2025-08-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

While working on a v1.0 file from the forum I notice that opening it in v1.1 shows multiple errors.

[coindispenser-roy.FCStd.zip](https://github.com/user-attachments/files/22009309/coindispenser-roy.FCStd.zip)
(remove the .zip extension)

Forum topic:
https://forum.freecad.org/viewtopic.php?t=99369
```
Pad011: Result has multiple solids: enable 'Allow Compounds' in the active body.
<Exception> No mirror plane reference specified
<Exception> No mirror plane reference specified
<Exception> No mirror plane reference specified
<Exception> No mirror plane reference specified
<Exception> No axis reference specified
<Exception> No axis reference specified
<Exception> No mirror plane reference specified
<Exception> No mirror plane reference specified
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Sketch013.ExternalGeometry missing element reference coindispenser_roy#Pad009 ;g1;SKT;:Hb69,E.Edge5
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Sketch013.ExternalGeometry missing element reference coindispenser_roy#Pad009 ;g2;SKT;:Hb69,E.Edge2
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Sketch013.AttachmentSupport missing element reference coindispenser_roy#Pad009 ;#390:2;FAC;:Hb69:4,F.Face3
PositionBySupport: AttachEngine3D: subshape not found Pad009.Face3
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Sketch014.ExternalGeometry missing element reference coindispenser_roy#Mirrored002 ;#403:5;:U;RFI;:H-b67:7,E;:Hb67,E.Edge21
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Sketch014.ExternalGeometry missing element reference coindispenser_roy#Mirrored002 ;#3a7:7;:U;XTR;:Hb6a:7,E;:H,E;:Hb67,E.Edge8
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Sketch014.AttachmentSupport missing element reference coindispenser_roy#Mirrored002 ;#3d0:5;:G;RFI;:H-b67:7,F;:Hb67,F.Face6
PositionBySupport: AttachEngine3D: subshape not found Mirrored002.Face6
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Joint005.Reference1 missing element reference coindispenser_roy#pal pal.;g47;SKT;:Hb73,E;:Hb74,E.Edge38
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Joint005.Reference1 missing element reference coindispenser_roy#pal pal.;g47;SKT;:Hb73,E;:Hb74,E.Edge38
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Joint004.Reference1 missing element reference coindispenser_roy#pal pal.;g48;SKT;:Hb73,E;:Hb74,E.Edge20
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Joint004.Reference1 missing element reference coindispenser_roy#pal pal.;g48;SKT;:Hb73,E;:Hb74,E.Edge20
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Joint001.Reference1 missing element reference coindispenser_roy#pal pal.;#4f8:18;FAC;:Hb73:4,F;:Hb74,F.Face14
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Joint001.Reference1 missing element reference coindispenser_roy#pal pal.;#4f8:18;FAC;:Hb73:4,F;:Hb74,F.Face14
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Joint.Reference1 missing element reference coindispenser_roy#pal pal.;#507:18;:G0;XTR;:Hb73:8,F;:Hb74,F.Face15
<PropertyLinks> PropertyLinks.cpp(515): coindispenser_roy#Joint.Reference1 missing element reference coindispenser_roy#pal pal.;#511;:U;XTR;:Hb73:7,V;:Hb74,V.Vertex6
```

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

oursland, 2025-08-27:
@drwho495 This issue has a file that can be used to reproduce the issue.

drwho495, 2025-08-27:
> @drwho495 This issue has a file that can be used to reproduce the issue.

Ok thanks, I'll fix it

drwho495, 2025-08-27:
please try: https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.08.27

i see no errors there

Roy-043, 2025-08-27:
That is strange because I have used that weekly.

drwho495, 2025-08-27:
> That is strange because I have used that weekly.

could you take a video?

Roy-043, 2025-08-28:
Issue reported on the forum that may be related:
https://forum.freecad.org/viewtopic.php?t=99391

drwho495, 2025-08-28:
> A video to record the opening of a file would be a bit strange IMO.
> 
> Opening the file in Safe Mode the errors are reduced to only (which is surprising):
> ```
> 10:15:12  <Exception> No mirror plane reference specified
> 10:15:12  <Exception> No mirror plane reference specified
> 10:15:12  <Exception> No mirror plane reference specified
> 10:15:12  <Exception> No mirror plane reference specified
> 10:15:12  <Exception> No axis reference specified
> 10:15:12  <Exception> No axis reference specified
> 10:15:12  <Exception> No mirror plane reference specified
> 10:15:12  <Exception> No mirror plane reference specified
> ```

Ok then it's probably related to settings (?)

What I might have you do is build a branch of my fork and I'll tweak things until it works for you

Roy-043, 2025-08-28:
Maybe the Dag View performs an analysis and while doing that finds additional problems?

davidrihtarsic, 2025-08-31:
I would like to confirm that I experienced a very similar problem when opening a file created in v1.0.2.  
In my case, opening the file in FreeCAD 1.1 dev caused the program to crash due to broken **External geometry references** inside a sketch.  
Console output showed messages like:

```
<PropertyLinks> PropertyLinks.cpp(492): Sketch001.ExternalGeometry missing element reference … Edge2 / Edge8
freecad: .../SketchObject.cpp:10162: Assertion `externalGeoRef.size() == objs.size()` failed.
```

As suggested by @drwho495, I tried the [weekly build 2025.08.27](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.08.27):  
With this build, the same file opens without crash.  

So I can confirm that the issue seems resolved in that version.

Now, I get similar error to @Roy-043:

```
<Exception> No axis reference specified
<Exception> No mirror plane reference specified
```
... however, the model is not broken and I've checked every Revolve, Groove, Mirror, PolarPattern, LinearPattern in the model and **reference planes** are defined... I'll dig deeper. 

Thanks for any suggestions in advanced!