FreeCAD/FreeCAD #23403: Using Toggle Transparency, resets the color of the object

belinski-roman, 2025-08-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The Toggle Transparency application resets the object's color if it has been assigned by the assigned material or using Random Color. If the color was assigned via the Property View, everything is working properly.

### Full version info

```shell
OS: Windows 10 build 19044
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Object center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/102.299/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-28:
I cannot reproduce this issue, works as expected.

[image]

```
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
Installed mods: 
  * manifest.json

```

maxwxyz, 2025-08-29:
I think porting or fixing https://github.com/FreeCAD/FreeCAD/issues/11210 will close this here too

closed 2026-03-09 by maxwxyz