FreeCAD/FreeCAD #23407: Sketcher: Make Internals visibility from behind

maxwxyz, 2025-08-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The internal sketcher faces have a different visibility from the backside. Even worse: the selection of a face is not visible from the back side.
In this example, the right internal face is selected but is not visible:

[image]

### Full version info

```shell
OS: macOS 13.7.8
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/72/1
Installed mods: 
  * freecad.gears 1.3.0
  * CfdOF 1.31.4
  * Defeaturing 1.2.6
  * Gridfinity 0.11.9
  * AddonManager 2025.7.17
  * Manipulator 1.6.0
  * fasteners 0.5.39
  * lattice2 1.0.0
  * Telemetry 1.0.2
  * CurvedShapes 1.0.13
  * Cables 0.2.1
  * FusedFilamentDesign 0.25.250
  * Curves 0.6.64
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-27:
@PaddleStroke FYI

FEA-eng, 2025-08-27:
> The internal sketcher faces have a different visibility from the backside.

The benefit of that is that normal direction of a sketch is clearly visible.