FreeCAD/FreeCAD #23416: Sketcher: Make internal breaks Part WB operations

maxwxyz, 2025-08-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It seems Sketcher "Make Internals" is not supported by Part WB.

In this example I extruded the intersecting internal face of a sketch.
The appearance is broken for the result.
The calculated shape is broken for the result.
It is not possible to use internal faces in Part.

[image]

File:

[PartInternalFaces.zip](https://github.com/user-attachments/files/22018213/PartInternalFaces.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1
Installed mods: 
  * manifest.json
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-28:
@PaddleStroke @kadet1090 FYI

FEA-eng, 2025-08-28:
Even with Make Internals disabled, all sketch-based operations in Part wb are affected:

[image]

I think that it should be a blocker.

maxwxyz, 2026-03-28:
IMO this is fixed with https://github.com/FreeCAD/FreeCAD/pull/28648

[image]

The original file fails in Part regardless of the Make Internals property.
@FEA-eng can you confirm?

Should Internal Faces be enabled by default?