FreeCAD/FreeCAD #23430: PD: More info when adding a Sketch when there is no active Body.

leoheck, 2025-08-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

More info when adding a Sketch when there is no active Body.

If labels are equal, I cannot differentiate bodies.
I would adopt a strategy like this (everywhere)

If `Label == Name` show just the `Label`.
If not, then show `Label (Name)`.

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git)
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * manifest.json
  * FreeCAD-render 2024.12.15
  * Assembly4 0.60.2-dev
  * sheetmetal 0.7.24
  * DynamicData 2.74.0
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * osh-autodoc-workbench 0.2.3
  * Assembly4p1 0.50.18
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-28:
If the user manually changes all the bodies to "Body" what is expected? The same applies to all dialogs where you select references (sketches, datums,...)

closed 2025-08-28 by maxwxyz

leoheck, 2025-08-29:
The user did not manually changed anything. There is a settings in the preferences where it allows the objects to be created with a label that doesn't have the annoying number that makes names uniques.

I just created 2 bodies. Didn't change the labels manually.