FreeCAD/FreeCAD #23434: PartDesign: Can’t select multiple internal faces of a sketch

pierreporte, 2025-08-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

With #22388, it is now possible to easily pad or pocket a subset of a sketch. However it is only allowed to select one internal face. It should be allowed to select more than one.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * sheetmetal-backup-before-zip-migration 0.7.22 (Disabled)
  * Curves-backup-before-zip-migration 0.6.63 (Disabled)
  * FusedFilamentDesign-backup-before-zip-migration 0.25.220 (Disabled)
  * fasteners 0.5.41
  * Curves 0.6.70
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.24
  * manifest.json
  * FusedFilamentDesign 0.25.250
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-08-28:
I can select and pad (if multiple solids are enabled) multiple faces. But only the first one has visibly dark highlight.

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

maxwxyz, 2025-08-28:
The entire draw style of those internal faces needs some rework, like https://github.com/FreeCAD/FreeCAD/issues/23407
Maybe also fixed with @tetektoza new selection system :)

wohltat, 2026-03-25:
Bug still present in 1.1 release

OS: Manjaro Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0.20260325 (Git shallow) AppImage
Build date: 2026/03/25 02:17:53
Build type: Release
Branch: (HEAD detached at 34a971666)
Hash: 34a9716668b1ddeb55b914f1c5be644826bdbbbf
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 76/122.918/1
Installed mods: 
  * Launcher
  * OpticsWorkbench 1.2.2
  * ShortCuts