FreeCAD/FreeCAD #23451: BIM: Arch_Roof is placed as a child object of Wall when it is Arch_Removed

semhustej, 2025-08-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**This issue happens with files, which were created in FreeCAD 1.0. 
I can not reproduce the issue with files created in current FreeCAD 1.1dev version**

Arch_Roof is placed as a child object of Wall (and hidden) when it is Arch_Removed from the wall.

Steps to reproduce:

- open sample file created in FC 1.0 (remove the zip extension)

[roof-testing-pol.FCStd.zip](https://github.com/user-attachments/files/22047862/roof-testing-pol.FCStd.zip)

- select the Roof object
- add one of the walls to the selection group
- use `Arch_Remove` command
- tho roof is removed from the Wall, but it is also placed in the wall container as child (this does not happen normally)

See behavior in the video:

https://github.com/user-attachments/assets/db24cc38-535c-4969-bd89-74b80aeb5255

I am not able to reproduce the issue in FC 1.0.0

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rsevat, 2025-08-29:
Extra context. If I restart FreeCAD. Select the Roof, Select the wall, change to BIM workbench, press the minus button. It correctly trims the wall. It does still print "$name has no solid". But it removes the wall. If I then try to do another wall, it fails again.

However if I close FreeCAD. Restart it again. And do the exact same thing as described above. I can trim 1 wall again. So if I restart FreeCAD 10 times I can trim all my walls which is a very hacky workaround.

I also tried FreeCAD 1.0 or 1.0.1, they show the same behavior.

semhustej, 2025-08-30:
It does not break the model. It can be worked around. So in that sense I would not call it a blocker.

But it is very confusing to the users, that one model behaves in a different way than another, with objects disappearing for no reason and a different project tree structure. I don't think 1.1 should be released with this bug.

Roy-043, 2025-08-30:
Oops, my bad. Caused by #22191.

closed 2025-09-01 by yorikvanhavre