FreeCAD/FreeCAD #23455: BIM: interactive sun position and ray visualization doesn't work

pgilfernandez, 2025-08-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

interactive sun position and ray visualization doesn't work (just macOS?)

### Full version info

```shell
Under macOS the just added feature "interactive sun position and ray visualization" doesn't work, it throws this error:

The pysolar module was not found. Unable to generate solar diagrams
Ladybug or Pysolar module not found. Cannot calculate sun position.

OS: macOS 12.7.6
Architecture: x86_64
Version: 1.1.0dev.43071 (Git)
Build date: 2025/08/25 18:08:30
Build type: Release
Branch: main
Hash: d85613b498881c47485a7d930807673203086756
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: MayaGesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/129/2
Installed mods: 
  * PieMenu 1.12.0
  * DynamicData 2.74.0
  * ProDarkThemePreferencePack 1.0.0
  * Nodes 0.1.36
  * freecad.gears 1.3.0
  * OpenTheme 2025.5.20
  * manifest.json
  * SelectorToolbar
  * ExtremeProDark 2.9.0
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pgilfernandez, 2025-08-29:
well, I guess FreeCAD must include `pysolar` package by default and not force users to install it manually... I guess testers and advanced users may be able to do it manually by themselves but the issue is open to devs to add this package in all platforms by default... doesn't it makes sense?

Thanks

semhustej, 2025-08-29:
@pgilfernandez 

I was able to install ladybug on Windows running this snippet in the python console. You might try replacing the `C:/....` path with the path to the FreeCAD installation on Mac. Note that I am not a developer and I am only guessing this.

```
import subprocess
subprocess.run([
"C:/Program Files/FreeCAD 1.0/bin/python.exe",
"-m", "pip", "install", "ladybug-core"
])
```

pgilfernandez, 2025-08-30:
> I was able to install ladybug on Windows running this snippet in the python console. You might try replacing the `C:/....` path with the path to the FreeCAD installation on Mac. Note that I am not a [...]

Thank you very much,  with your snippet I was able to install `ladybug-core` and make it working. I just changed yours into:
```
import subprocess
subprocess.run([
"/Applications/FreeCAD.app/Contents/Resources/bin/python",
"-m", "pip", "install", "ladybug-core"
])
```

I'm still getting an error when changing settings but it is easy to solve:
```
16:31:10  pyException: Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/BIM/ArchSite.py", line 1150, in onChanged
    self.updateSunPosition(vobj)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/BIM/ArchSite.py", line 1436, in updateSunPosition
    vo.ArrowType = "Arrow"
    ^^^^^^^^^^^^
<class 'AttributeError'>: 'PartGui.ViewProviderPartExt' object has no attribute 'ArrowType'
```

So, yes, this new feature is cool and works but it depends on a package not installed by default, as for UX I would recommend to make FreeCAD bundle this package as default or warn users with an intrusive popup if they activate the **Solar Diagram** and `ladybug-core` is not installed (I forget about `pysolar` as you are already suggesting the other one is recommended).
I wonder if within that popup a simple button for installing this package could be added and this way the "circle would be closed".

BTW, thanks for your help and I hope my comments help to make it easier to people facing my "normal user" problems...  =)
Cheers

pgilfernandez, 2025-08-30:
I'm closing it as there is a way to manually solve the problem but I still suggest devs to add a popup or/and bundling `ladybug-core` into vanilla FreeCAD.

Thanks!

closed 2025-08-30 by pgilfernandez

semhustej, 2025-08-30:
@pgilfernandez thanks for posting the code for installing on Mac, someone might find it useful.

I can reproduce the error you are getting: https://github.com/FreeCAD/FreeCAD/pull/22516#issuecomment-3237825238. furgo16 is working on this feature, these are the 'alpha' bugs that appear when a new code is implemented. It is good that the feature is tested on Mac, Linux and Windows.