FreeCAD/FreeCAD #23461: "Wall has no solid" error when subtracting roof

rsevat, 2025-08-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I've been trying to ages now to get the walls to fit my roof. For some reason it fails for the walls that I've added on the second floor. Except for some it randomly succeeded. It throws the error "$name has no solid".

Someone else managed to reproduce it with my file, but not with a new file. It seems to be a bug.

https://www.reddit.com/r/FreeCAD/comments/1n2vebg/wall_has_no_solid_error_when_subtracting_roof/

I've attached the model to the github issue.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.3 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rsevat, 2025-08-29:
Extra context. If I restart FreeCAD. Select the Roof, Select the wall, change to BIM workbench, press the minus button. It correctly trims the wall. It does still print "$name has no solid". But it removes the wall. If I then try to do another wall, it fails again.

However if I close FreeCAD. Restart it again. And do the exact same thing as described above. I can trim 1 wall again. So if I restart FreeCAD 10 times I can trim all my walls which is a very hacky workaround.

I also tried FreeCAD 1.0 or 1.0.1, they show the same behavior.

I've recorded the behavior in this video:
https://github.com/user-attachments/assets/a61fc475-ce9a-4420-b7cf-76d5a9d73001

So if you restart FreeCAD again, you can remove 1 wall again. So I managed to work around this bug for now by restarting FreeCAD 10 times and trimming 1 wall at a time.

rsevat, 2025-08-29:
All walls trimmed by restarting FreeCAD for every wall:: [image]

semhustej, 2025-08-30:
Note: it is currently not possible to test the issue with the current 1.1dev because of this issue: https://github.com/FreeCAD/FreeCAD/issues/23451

I can reproduce the issue on the OP model. I think it is some weird interaction of Wall and Roof subtraction.
I am not able to recreate this issue on my own though.

-------

I was able to extract just few parts from the OP model and recreate the issue more reliably. It is linked to using the Undo tool:

- open the file (remove the zip extension)

[roof-testing-reddit .FCStd.zip](https://github.com/user-attachments/files/22057234/roof-testing-reddit.FCStd.zip)

- select the `Roof` and `Wall A` and use `Arch_Remove` tool
- select the `Roof` and `Wall B` and use `Arch_Remove` tool
- use `Std_Undo` twice to get to the original state
- you might need to repeat the above three steps again
- select the `Roof` and `Wall A` and use `Arch_Remove` and the bug shows up `wall A has no solid`

See the behavior in the video:

https://github.com/user-attachments/assets/9186504e-f128-4b2d-b66e-38c0e2212bbb

I am not sure about that, but I think similar issue, that behaves in a different way, was encountered by this user: https://forum.freecad.org/viewtopic.php?t=98718

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * manifest.json
  * Reporting
  * Telemetry 1.0.2

```

rsevat, 2025-08-30:
After having continued, The door in the above screenshot suddenly has decided to render correctly and create a hole in the wall. The other doors however don't render correctly yet.

And it's still spamming these errors: 
`17:48:18  gang zuid001 has no solid
17:48:18  gang zuid001 has no solid
17:48:21  gang zuid001 has no solid
17:48:57  gang zuid001 has no solid
17:49:22  gang zuid001 has no solid
17:49:34  gang zuid001 has no solid
17:49:54  gang zuid001 has no solid
17:52:35  gang zuid001 has no solid
17:52:35  gang zuid001 has no solid
17:58:36  gang noord has no solid
17:58:56  gang noord has no solid
18:00:15  gang zuid001 has no solid
18:00:15  gang noord has no solid
18:00:25  gang zuid001 has no solid
18:00:36  gang zuid001 has no solid
18:02:13  gang noord has no solid
18:02:22  gang noord has no solid
18:02:29  gang noord has no solid
18:02:41  gang noord has no solid`

I'm hoping that over time they suddenly decide to render correctly (or if the no solid bug gets fixed that they render)

rsevat, 2025-08-30:
I managed to get all doors open by restarting FreeCAD a lot. Selecting a wall with a door that doesn't render properly. Pressing "recompute object" for that specific wall. And then it renders right. Same fix pretty much as for the roof.

So somehow upon startup it isn't in the 'bugged' state yet. But after rendering, it then does get in the bugged state.

No worries @semhustej I understand. Thanks a lot for your effort and thinking along. I will try the development version as soon as that MR gets merged to see what happens.

I'm almost done with the house. I'll continue with this workaround of restarting FreeCAD a lot. Main thing left is that I need to generate the sections/tech drawings so I can request a building permit for the renovation of our house.

Fingers crossed that there is no corruption in the actual model and that it can be solved with a bugfix that hopefully makes it into 1.1.

paullee0, 2025-09-04:
Not sure what is wrong, normal to me here.

- Adding Roof objects manually to Subtraction of each Wall object, recompute the latter, done.

[model_rev.FCStd.Txt](https://github.com/user-attachments/files/22161937/model_rev.FCStd.Txt)

[image]

[image]

[image]

rsevat, 2025-09-10:
I can indeed not reproduce the issue anymore with FreeCAD_weekly-2025.09.10-Linux-x86_64-py311.AppImage with the model that is attached to this ticket.

I however still have some versions of the model in which I can still reproduce the issue, even on the latest 10 Sept weekly build.

I've attached a model in which I can still reproduce it on the latest development build.

[huis model - wall has no solid.zip](https://github.com/user-attachments/files/22263108/huis.model.-.wall.has.no.solid.zip)

`OS: Ubuntu 24.04.3 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/107.975/1
`