FreeCAD/FreeCAD #23462: New Transform Tool usually doesn't show xyz/rotation tool in 3D view if a clone is selected

dekay, 2025-08-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Please see attached images and sample file illustrating the problem. The stuff on the right of this image represent my cut plan for the pinball machine I'm designing on the left. So all the elements on the right are clones. These clones aren't contained within a body but just lumped within a Part. The problem I'm having is 9 times out of ten, I'll try to use the transform tool on those clones but the dragger tool that lets me manipulate the clone doesn't appear in the 3D view. One time out of ten it will appear and work well. I can't figure out the pattern. It just seems random. I never have the problem with the on-screen thingie appearing for non-clones.

The attached images are from the full drawing that is 23MB in size. I have cut out most of the stuff in the design except for the wood parts in the cut plan to make this more manageable. All of this was designed using the Part Design tool and the clones with that tool's Clone feature. They show the correct active object selected and the Transform tool activated but no on screen dragger is shown. Again, the odd time this will show up but most of the time it won't.

The new transform tool is great but this bug really slows me down and probably others as well so fingers crossed it can be considered relatively high priority. The workaround of editing by changing the numbers in the task panel is very clunky.

Any ideas on this, @kadet1090?

[TransformBugReport.zip](https://github.com/user-attachments/files/22050041/TransformBugReport.zip)

[image]

[image]

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/20 00:00:55
Build type: Release
Branch: main
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: C/Default (C)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Object center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/110.786/1
Installed mods: 
  * Curves 0.6.60
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-08-29:
@dekay does the dragger show in 1.0 for clones?

dekay, 2025-08-29:
> [@dekay](https://github.com/dekay) does the dragger show in 1.0 for clones?

@maxwxyz Yes. I forgot to mention this. The dragger showed 100% of the time when I did the design initially in 1.0.0 and now on 1.0.2 as well. It would be fair to consider this a regression, I think.

kadet1090, 2025-08-30:
Ok, I see. Your file uses Base Features that are moved outside of bodies (I presume you dragged them out of bodies manually?). That is something that should not really be supported - Base Feature should be always placed in body and body should always have one Base Feature. For some reason the code allowed it to be moved out and by pure luck as none code path that tried to access information from the Body it worked well. 

I'll think how to handle it.

dekay, 2025-08-30:
@kadet1090 Yes, I did drag them out of the bodies manually. All the bits on those two 4'x8' panels on the right are clones and doing it the way I did made the tree a lot less cluttered. Having a body with just a single clone in it and nothing else didn't seem at the time to be useful. Freecad seemed happy to let me do it and I didn't lose any apparent functionality in the 1.0 version when I did it this way, so that's what I did as a Freecad newbie.

kadet1090, 2025-08-30:
Yep, software let you did it, it seems reasonable so you had no way to know that it should not be supported. Given how it worked I'll try to make some special case to handle scenarios like yours (as you are probably not the only one who did it) and block features from being moved from the body in new files as we have multiple problems due to this including random segfaults.

dekay, 2025-08-31:
@kadet1090 I went ahead and tried FreeCAD_weekly-2025.08.27-Linux-x86_64-py311.AppImage, knowing that it probably wouldn't fix my issue. It actually makes matters much worse. Selecting Edit->Transform with a close selected immediately crashes Freecad. So I ran it under gdb and got this...
```<Exception> No direction reference specified
<Exception> No direction reference specified
Extension is not a python addable version: 'App::SuppressibleExtension'
Extension is not a python addable version: 'Part::PreviewExtension'
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3e540) [0x7ffff1c3e540]
#1  0x7ffed8dc4a81 in PartDesignGui::ViewProvider::setEdit(int) from /tmp/.mount_FreeCAIbOkOb/usr/lib/PartDesignGui.so+0xb1
#2  0x7ffed8de9df7 in PartDesignGui::ViewProviderBase::setEdit(int) from /tmp/.mount_FreeCAIbOkOb/usr/lib/PartDesignGui.so+0x77
#3  0x7ffff7d76461 in Gui::ViewProvider::startEditing(int) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0x11
#4  0x7ffff7d92275 in Gui::ViewProviderDragger::startEditing(int) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0x15
#5  0x7ffed8dc5e1f in PartDesignGui::ViewProvider::startEditing(int) from /tmp/.mount_FreeCAIbOkOb/usr/lib/PartDesignGui.so+0x1f
#6  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so(+0x1cd22aa) [0x7ffff78d22aa]
#7  0x7ffff78c36c3 in Gui::Document::trySetEdit(Gui::ViewProvider*, int, char const*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0x393
#8  0x7ffff78c3131 in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0x21
#9  0x7ffff799c801 in StdCmdTransformManip::activated(int) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0xb1
#10  0x7ffff7980c4a in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0x23a
#11  0x7ffff79809c0 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0x150
#12  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7ffff231684a]
#13  0x7ffff2dca8c7 in QAction::triggered(bool) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Gui.so.6+0x47
#14  0x7ffff2dcddc6 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Gui.so.6+0xf6
#15  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6(+0x3cf445) [0x7ffff35cf445]
#16  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6(+0x449332) [0x7ffff3649332]
#17  0x7ffff37c3864 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6+0xa64
#18  0x7ffff37b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6+0x88
#19  0x7ffff37b4cbd in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6+0xdad
#20  0x7ffff7922ea8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0x78
#21  0x7ffff235fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Core.so.6+0x17a
#22  0x7ffff37bbe7c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6+0x1ec
#23  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6(+0x5bee09) [0x7ffff37bee09]
#24  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6(+0x5c5440) [0x7ffff37c5440]
#25  0x7ffff37b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Widgets.so.6+0x88
#26  0x7ffff7922ea8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0x78
#27  0x7ffff235fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Core.so.6+0x17a
#28  0x7ffff29d1970 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Gui.so.6+0x810
#29  0x7ffff2a31e14 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Gui.so.6+0xb4
#30  /tmp/.mount_FreeCAIbOkOb/usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x58f90) [0x7fffedc8ff90]
#31  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/./libglib-2.0.so.0(+0x5b646) [0x7fffeed08646]
#32  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/./libglib-2.0.so.0(+0x5e817) [0x7fffeed0b817]
#33  /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fffeed0bf2e]
#34  0x7ffff20f6b12 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Core.so.6+0x72
#35  0x7ffff239747c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Core.so.6+0x25c
#36  0x7ffff23977da in QCoreApplication::exec() from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libQt6Core.so.6+0x9a
#37  0x7ffff784bcd8 in Gui::Application::runApplication() from /tmp/.mount_FreeCAIbOkOb/usr/bin/../lib/libFreeCADGui.so+0xc88
#38  /tmp/.mount_FreeCAIbOkOb/usr/bin/freecad(+0x9c35) [0x55555555dc35]
#39  /usr/lib/libc.so.6(+0x27675) [0x7ffff1c27675]
#40  /usr/lib/libc.so.6(__libc_start_main+0x89) [0x7ffff1c27729]
#41  /tmp/.mount_FreeCAIbOkOb/usr/bin/freecad(+0x8d59) [0x55555555cd59]
[Inferior 1 (process 20862) exited with code 01]
```