FreeCAD/FreeCAD #23481: BIM: BIM Preferences override FreeCAD Preferences (macOS)

pgilfernandez, 2025-08-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If you use any workbench and want to edit FreeCAD preferences you just open them like the following:

[image]

But if you open BIM workbench, then they are overridden by BIM Preferences:

[image]

If you close the BIM Preferences dialog, then the normal FreeCAD ones are shown but not always... sometimes you need to open/close the BIM ones a few times to get FreeCAD Preferences to finally show up...

### Full version info

```shell
OS: macOS 12.7.6
Architecture: x86_64
Version: 1.1.0dev.43071 (Git)
Build date: 2025/08/25 18:08:30
Build type: Release
Branch: main
Hash: d85613b498881c47485a7d930807673203086756
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: MayaGesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/129/2
Installed mods: 
  * PieMenu 1.12.0
  * DynamicData 2.74.0
  * ProDarkThemePreferencePack 1.0.0
  * Nodes 0.1.36
  * freecad.gears 1.3.0
  * OpenTheme 2025.5.20
  * manifest.json
  * ExtremeProDark 2.9.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-09-02:
> [@JohnOCFII](https://github.com/JohnOCFII) would you mind confirming on macOS ?

Confirmed.  Exactly as described by the OP.

```
OS: macOS 15.6.1
Architecture: arm64
Version: 1.1.0dev.43185 (Git)
Build date: 2025/08/31 23:44:02
Build type: Unknown
Branch: main
Hash: b89870384198cd0d779706366c9354bce4d457b9
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.8.23
  * manifest.json
  * Telemetry 1.0.2
```

Roy-043, 2025-09-02:
Which BIM dialog is displayed when you click the faulty menu item?

[image]

[image]

Roy-043, 2025-09-19:
Should be fixed with #24014.

Roy-043, 2025-10-23:
Could somebody with a Mac verify if the issue has been fixed in the current weekly? Thanks.