FreeCAD/FreeCAD #23504: Part Design: Using the dragger causes changes to be accepted on "Esc" key

ryankembrey, 2025-08-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using the new interactive controls in the Part Design, if the user interacts with the dragger in the 3D view, the changes will be accepted if the "Esc" key is pressed. 
As far as I can tell, this only occurs when the dragger is interacted with. This issue does not occur when the fields of the task panel are edited manually.

Have tested on 1.1-dev (occuring) and 1.0.1 (non-occuring as draggers weren't implemented yet)

## Demo of Issue

### Full version info

```shell
OS: Fedora Linux 41 (Sway) (sway/sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.43160 (Git)
Build date: 2025/08/31 11:41:02
Build type: Unknown
Branch: dimensions
Hash: c5c2af2ba125f87969d5ededc493154f896881df
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * OpenTheme 2025.5.20
  * Render 2024.12.15
  * DFM
  * manifest.json
  * sheetmetal 0.7.24
  * CfdOF 1.31.6
  * fasteners 0.5.41
  * Telemetry 1.0.2
  * AddonManager 2025.8.17dev
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

captain0xff, 2025-09-01:
As I had guessed this issue is not really caused by the draggers rather they just intensify it's effects. You can reproduce the same problem in 1.0.x by clicking in the viewport and then pressing escape.
I think what is happening here is that once you are clicking the viewport, the task panel is losing focus and so the keyevents are not reaching to it. If you click in the task panel after using the draggers then everything works as expected.
This problem can be also demonstrated through the enter key. Once, someone clicks on the viewport, pressing enter doesn't confirm the operation anymore.

yusufm, 2025-10-02:
Verified that this is fixed in (confirmed that it was broken in 9/23 build):

OS: macOS 26.0.1
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2