FreeCAD/FreeCAD #23508: BIM: Error message shows up when enabling Solar Diagram

semhustej, 2025-08-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Priority: low

When enabling Solar Diagram of a site object for the first time a FreeCAD is restarted, an error (red) message shows up.

Steps to reproduce:

- open new file
- create `Arch_Site` object
- enable `Solar Diagram` property of the Site object
- error message (red) shows up:

`INFO:ladybug:Successfully imported Ladybug plugin: ladybug_geometry`

See the behavior in the video:

https://github.com/user-attachments/assets/aebc9f20-c6c7-4e81-8046-f323273f11d3

This happens only once after FreeCAD is restarted.

I am not able to reproduce with FC 1.0.2

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * manifest.json
  * Reporting
  * Telemetry 1.0.2
```

### Subproject(s) affected?

BIM

### Anything else?

I think the message should be an information message - black color.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-08-31:
@furgo16 i think it's related to the last Solar Study changes.

ErtugrulEge, 2025-08-31:
I can confirm 

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1

furgo16, 2025-08-31:
I can also reproduce. The strange thing is that `ladybug` was already being imported before PR https://github.com/FreeCAD/FreeCAD/pull/22516. The only difference is that that PR additionally imports `ladybug`'s `location` module.

In any case, that message is being printed by `ladybug`'s logger. I do not know why the logger level is set to INFO upstream, when generally it's set to warning or error. I don't know either why the info level is being interpreted as an error by FreeCAD. 

A possible solution would be to either find out how to set the log level at the API level, or to intercept `ladybug`'s logger and set its level to error or warning. @tritao, as a Python expert, what are your thoughts?