FreeCAD/FreeCAD #23511: Export to STEP format does not work since version 1.0.2

wiserst, 2025-08-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Since version 1.0.2 I am unable to create a STEP file from an imported STL file using export. In the previous version (1.0.1) it worked correctly with the same procedure.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Czech/Czech Republic (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Manipulator 1.6.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-01:
Can you reproduce in the latest dev version in safe mode?

wiserst, 2025-09-02:
I tried the development version OS: Windows 10 build 19045 Architecture: x86_64 Version: 1.1.0dev.43071 (Git) Build date: 2025/08/25 18:08:30 run in safe mode (freecad.exe --safe-mode).

Same procedure, same result. Export doesn't work, no file is saved.

wiserst, 2025-09-02:
I tried the same procedure in older versions (1.0.0, 1.0.1) and it doesn't work there either. I think it worked in older version 0.21, but I don't have that anymore.

closed 2025-09-02 by maxwxyz