FreeCAD/FreeCAD #23514: Core: Expression Editor 'Reset' button name

semhustej, 2025-08-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature request**

Expression editor has `Reset` button. This button cancels the changes user did in the editor and closes the dialogue. This behavior is more consistent with the `Cancel` button. I think it is widely understood, that Reset button only resets the field to the original value and does not close any dialogue.

I think the `Reset` button should be renamed to `Cancel` to more clearly convey to the user what it does.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * manifest.json
  * Reporting
  * Telemetry 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-09-01:
Oh, I see. Thanks for the explanation.

In that case, I don't think the `Reset` button should close the Expression Editor. It should just reset the value. I understand it might be useful for experienced users as it saves them one click, but changing the value of the expression should be confirmed by the user in my opinion.

ChrisRozario-Aero, 2026-10-18:
Hi, I'd like to contribute to this issue.

After reading the discussion, I understand that the Reset button converts the current expression into its evaluated fixed value, rather than cancelling the dialog.

Before implementing a change, could the maintainers clarify the intended behavior and preferred button label? "Convert to fixed value" seems explicit, while "Evaluate" is more concise.

I also noticed issue #32604, which discusses applying expressions as fixed values from the Property View. Should this naming improvement be handled separately or coordinated with that work?

Thank you!