FreeCAD/FreeCAD #23518: Pressing escape to close the expression editor also closes the task panel

davidgilkaufman, 2025-08-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Create an editable object (tested with Pad, CAM Profile). Double click to open its panel
2. Find an input field that can be edited as a formula. Open the formula editor
3. Press escape to close the formula editor

Problematic behavior: instead of simply closing the formula editor, pressing escape also closes the task panel.

I don't understand freecad's UI framework well enough to track down this problem myself, but fwiw I have also been able to trigger it from other dialogs, so I believe it is a problem with the task panel's key listener being overzealous, not the expression editor specifically failing to consume a key event.

I can avoid this behavior my focusing a different application's window between steps 2 and 3, refocusing the dialog (but not freecad's main window!), and then pressing escape. I don't know how broadly applicable this workaround is (I run wayland and a tiling window manager, and one or both of these may be required to execute this workaround) but imo it further supports my diagnosis that the task panel's key listener is overzealous.

Note versions: I tried to reproduce this on 1.0.1 (git 39285) and failed. I have been encountering this on my development branch based off of vaguely recent `main`. I'm certain it is not related to my changes; 
I am making exclusively CAM changes and have reproduced the bug outside of CAM.

### Full version info

```shell
OS: Arch Linux (wayland)
Architecture: x86_64
Version: 1.1.0dev.41396 +19 (Git)
Build date: 2025/08/31 16:25:35
Build type: Release
Branch: tool_controller_edit
Hash: e8aecc771670f814678646a13f925aeea5d3ae02
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.0, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/101.6/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-08-31:
@davidgilkaufman, if you launch _FreeCAD_ with:

```bash
$ QT_QPA_PLATFORM=xcb ./FreeCAD
```

does this problematic behavior remain?

jffmichi, 2025-09-03:
I think the problem can be outlined as follows:

User presses escape
Qt closes the expression dialog as expected
MainWindow becomes active
that triggers an `ActivationChange`  event and calls `MainWindow::setActiveSubWindow`, `MainWindow::setActiveWindow` and `view->setFocus()` for the active MDI view, i.e. `View3DInventorViewer`
Qt triggers a `KeyRelease` event for escape which gets sent to the now focussed view
the `QuarterWidget` in the view forwards this event to the inventor scene graph
the `KeyRelease` event ends up in `ViewProvider::eventCallback` resulting in a call to `Document::resetEdit()`
this ultimately closes/deletes the now active task dialog

There are a lot more calls to different functions inbetween but that's the rough chain of events. I have no idea where exactly this should be fixed. The call to `view->setFocus()` was the only way I could find to get the correct MDI view active as outlined in the commend in 6de1358.

ipatch, 2026-11-15:
yes i believe this issue is still relevant.