FreeCAD/FreeCAD #23536: Start: Modernize Start page example files (so they don't report migrating to v1.1 in the Report View upon opening them)

luzpaz, 2025-09-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Start FC
2. From the Start page open ArchDetail example

Result:
```
09:06:44  v1.1, Wire006, migrated view properties
09:06:44  v1.1, Array, added hidden property 'PlacementList'
09:06:44  v1.1, Array001, added hidden property 'PlacementList'
09:06:44  v1.1, Array002, added hidden property 'PlacementList'
09:06:44  v1.1, Wire011, migrated view properties
09:06:44  v1.1, Wire001, migrated view properties
09:06:44  v1.1, Vented T profile, migrated view properties
09:06:44  v1.1, Wire014, migrated view properties
09:06:44  v1.1, Wire045, migrated view properties
09:06:44  v1.1, Wire046, migrated view properties
09:06:44  v1.1, Wire028, migrated view properties
09:06:44  v1.1, Wire031, migrated view properties
09:06:44  v1.1, Wire033, migrated view properties
09:06:44  v1.1, Wire030, migrated view properties
09:06:44  v1.1, Wire029, migrated view properties
09:06:44  v1.1, Wire032, migrated view properties
09:06:44  v1.1, Wire039, migrated view properties
09:06:44  v1.1, Wire020, migrated view properties
09:06:44  v1.1, Wire037, migrated view properties
09:06:44  v1.1, Wire021, migrated view properties
09:06:44  v1.1, Wire040, migrated view properties
09:06:44  v1.1, Wire043, migrated view properties
09:06:44  v1.1, Wire041, migrated view properties
09:06:44  v1.1, Wire042, migrated view properties
09:06:44  v1.1, Wire044, migrated view properties
09:06:44  v1.1, Wire, migrated view properties
09:06:44  v1.1, Wire047, migrated view properties
09:06:44  v1.1, Wire050, migrated view properties
09:06:44  v1.1, Wire054, migrated view properties
09:06:44  v1.1, Wire056, migrated view properties
09:06:44  v1.1, Wire055, migrated view properties
09:06:44  v1.1, Wire057, migrated view properties
09:06:44  v1.1, Wire053, migrated view properties
09:06:44  v1.1, Wire061, migrated view properties
09:06:44  v1.1, Wire063, migrated view properties
09:06:44  v1.1, Wire169002010, migrated view properties
09:06:44  v1.1, Wire067, migrated view properties
09:06:44  v1.1, Wire066, migrated view properties
09:06:44  v1.1, Wire068, migrated view properties
09:06:44  v1.1, Wire070, migrated view properties
09:06:44  v1.1, Wire071, migrated view properties
09:06:44  v1.1, Wire069, migrated view properties
09:06:44  v1.1, Wire073, migrated view properties
09:06:44  v1.1, Wire072, migrated view properties
09:06:44  v1.1, Wire074, migrated view properties
09:06:44  v1.1, Wire105, migrated view properties
09:06:44  v1.1, Wire105 (2D), added 'ForceCompound' property
09:06:44  v1.1, Wire078, migrated view properties
09:06:44  v1.1, window profile Fleetwood T04 - fixed001, added 'ForceCompound' property
09:06:44  v1.1, Wire083, migrated view properties
09:06:44  v1.1, Wire091, migrated view properties
09:06:44  v1.1, Wire084, migrated view properties
09:06:44  v1.1, Wire085, migrated view properties
09:06:44  v1.1, Wire090, migrated view properties
09:06:44  v1.1, Wire092, migrated view properties
09:06:44  v1.1, Wire089, migrated view properties
09:06:44  v1.1, Wire086, migrated view properties
09:06:44  v1.1, Wire106, migrated view properties
09:06:44  v1.1, Wire106 (2D), added 'ForceCompound' property
[... 130 lines cut ...]
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1507/2.19792
Installed mods: 
  * sheetmetal 0.7.24
  * manifest.json
  * Telemetry 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-09-01:
* @furgo16 does #23524 fix the BIMExample error mentioned above ?

Roy-043, 2025-09-01:
Please recheck once #23497 is merged.

Roy-043, 2025-10-24:
Your question is valid. But it can be applied to FreeCAD as a whole. Keeping everything up-to-date is simply not possible.