FreeCAD/FreeCAD #23537: [OpenSCAD] [Toolbar] Cannot find icon when used in custom toolbar

r5to, 2025-09-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Adding tools from OpenSCAD to global custom toolbar makes it **fail to load icons for those tools on FreeCAD start** with warning:
Cannot find icon: OpenSCAD_[...]

[image]

It makes those icons not loadded also in OpenSCAD workbench itself. It is the same for every OpenSCAD tool added to custom toolbar:

[image]

Activation of "Auto-load" from Workbenches preferences for OpenSCAD doesn't make any difference.

I have tested every category of tools i have: standard tools + installed workbenches from Addon Manager.
Looks like that some workbenches have prevention to load tools to toolbar before start them in FreeCAD, some loads fine without restrictions:

[image]

Load those workbenches manually after FreeCAD starts (or checking "Auto-load" on startup) makes them to appear in toolbar, but at the end of it, which **does not equal to the order from toolbar editor**:

[image]

**Would be nice to reserve position for those not loaded icons for workflow reasons... with just text name of tool or something.**

In the case of OpenSCAD, it **fails too load those icons for entire FreeCAD session** and maybe "Auto-load" for OpenSCAD just doesn't work.

Working workaround for this icon load fail for OpenSCAD:
Choose OpenSCAD as startup default workbench in workbenches preferences.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * 3DfindIT 1.2.0
  * boltsfc 2022.11.5
  * Cables 0.2.1
  * Curves 0.6.70
  * Defeaturing 1.2.6
  * dodo 1.0.1
  * DynamicData 2.74.0
  * fasteners 0.5.41
  * freecad.gears 1.3.0
  * InventorLoader 1.5.1
  * kicadStepUpMod 11.7.3
  * manifest.json
  * Manipulator 1.6.0
  * MeshRemodel 1.10.38
  * Nodes 0.1.36
  * offline-documentation 1.0.0-alpha
  * OpenTheme 2025.5.20
  * ose-piping
  * Render 2024.12.15
  * SearchBar 1.5.0
  * sheetmetal 0.7.24
  * ThreadProfile 1.96.0
  * workfeature-macro
```

### Subproject(s) affected?

OpenSCAD

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-09-01:
@KeithSloan would you mind confirming this ?

closed 2025-09-01 by maxwxyz

relikd, 2025-12-06:
Is not specific to OpenSCAD. Came here to open the same issue. I am trying to import the `CAM_SelectLoop` action into the global toolbar. The icon shows fine when I add it to the toolbar – but does not load after I restart FreeCAD. Regardless of auto-load workbench setting. Additionally, the icon will be broken on the actual CAM workbench too. It seems the icon is only loaded on start and then never updated, even if the workbench is opened for the first time (in the session).

Fun fact, the `CAM_SelectLoop` action works fine even if I disable the workbench completely. (which I do appretiate in this case, because I dont want the workbench to show up in the workbench dropdown as this is the only feature I use.