FreeCAD/FreeCAD #23540: TechDraw: KeepLabel Option Not Working

WandererFan, 2025-09-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

original report: https://forum.freecad.org/viewtopic.php?t=99379

With the new handling of view frame (only appear on mouse-over), the KeepLabel property on views no longer works.

The Caption field is the preferred method of having a persistent text attached to the view.   Perhaps we should do something like:
if KeepLabel and Caption.empty()
    Caption = Label

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.43129 +1 (Git)
Build date: 2025/08/28 23:10:19
Build type: Unknown
Branch: ISOLineSpacing
Hash: 1f709614df349c733d7a4e4049fb1f59685b98cc
Python 3.10.12, Qt 6.9.1, Coin 4.0.0, Vtk 7.1.1, boost 1_74, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.8.0
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/118.229/1.19792
Installed mods: 
  * fasteners 0.5.21
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ryankembrey, 2025-09-02:
In my opinion, any text that is used to permanently label the view should be in the caption field. 

I think the suggestion @WandererFan gives is good. 

I read on the forum thread linked that the border padding is too large? What would be a better size? Half of the current?

WandererFan, 2025-09-03:
I'm a bit conflicted on this one, so I would not default this to true.   If it is true and this fix is in place, you'll get "View001" as a persistent caption on the page and on any exports.   I don't know how we'd tell that the caption should be shown because it contains something important vs being filled with the view label - if Label == Caption then do something?

WandererFan, 2025-09-03:
The more I think about using the Caption as something other than a caption, the less I like it.  There really isn't any need to clutter the display with automagic caption text since we now have a cool mouse-over feature that provides the same information.  

I would be perfectly happy with deprecating KeepLabel as I have never used it much.  Keeping the current main behaviour would be fine too.  

There are other demands on Caption to consider as well (issue #21908).

@FreeCAD/cad-working-group or @FreeCAD/design-working-group might have an option.

MisterMakerNL, 2025-09-04:
The section caption is a logical thing, but this also is necessary for detailed views. Is there a solution for this?

meprakarsa, 2026-03-05:
> Your weekly is 3 months old, can you try in the latest weekly?

sorry missed your comments, i assume you were replying to me...? I just downloaded FreeCAD 1.1 RC3, and the "keep label" checkbox of the techdraw view is still the same: unusable.

I will try to download the most current weekly build too.

closed 2026-08-27 by maxwxyz