FreeCAD/FreeCAD #23559: Overlay: Ctrl+Down shortcut disables Esc shortcut

furgo16, 2025-09-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In overlay mode, once the https://wiki.freecad.org/Std_DockOverlayToggleBottom command has been run once via the <kbd>Ctrl</kbd>+<kbd>Down</kbd> keyboard shortcut, the <kbd>Esc</kbd> keyboard shortcut is disabled. Pressing <kbd>Esc</kbd> no longer has any effect. To reenable it, overlay mode needs to be exit and reentered.

Steps to reproduce:

1. Start FreeCAD in Safe Mode
2. In Part Design, create a sketch
3. Double-click the sketch to enter edit mode
4. Press <kbd>Esc</kbd>
5. Sketch edit mode is cancelled => **PASS**
6. Press <kbd>F4</kbd> as a shortcut to activate overlay on all panels
7. Double-click the sketch to enter edit mode
8. Press <kbd>Esc</kbd>
9. Sketch edit mode is cancelled => **PASS**
10.Press <kbd>Ctrl</kbd>+<kbd>Down</kbd> to hide the bottom panel (Report view)
11. Double-click the sketch to enter edit mode
12. Press <kbd>Esc</kbd>
13. Sketch edit mode is NOT cancelled => FAIL
14. From this point on, <kbd>Esc</kbd> keypresses have no effect => **FAIL**

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.43236 (Git) Snap 1689
Build date: 2025/09/01 22:25:55
Build type: Release
Branch: main
Hash: 6c49ff9b5b8232f92fe4d3146acb0800052ba554
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Telemetry 1.0.1
  * Launcher
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-11-02:
Still very much relevant

FEA-eng, 2026-05-03:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260421 (Git shallow)
Build date: 2026/04/21 23:21:45
Build type: Release
Branch: (HEAD detached at 449725202)
Hash: 449725202d0ab581516aacad59d62cdce80db82a
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```