FreeCAD/FreeCAD #23574: Sketcher: v 1.1 crashes when I change the dimension of an offset.

tedology, 2025-09-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm a new FreeCAD user, so I'm not sure if this is a known issue or just new to v 1.1.

I run a Mac.

When I create an offset over constructed lines and attempt to set it as a constraint, I try to change the dimension of the offset and when I hit RETURN, the program crashes.

### Full version info

```shell
OS: macOS 15.6
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 72/92/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

matthiasdanner, 2025-09-04:
I think this is a duplicate of #23628

maxwxyz, 2025-09-05:
Closing this if there is not more info

closed 2025-09-05 by maxwxyz