FreeCAD/FreeCAD #23576: Sketcher : Arc offset dimension constraint doesn't work properly

ErtugrulEge, 2025-09-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everybody!
Steps to reproduce:
1.Draw a rectangle
2.Make an arc offset
3.Dimension one line of the offset

https://github.com/user-attachments/assets/af92e49a-ae9a-422f-8070-a34f1c6a7073

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * manifest.json
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-03:
Please provide an example file.
@PaddleStroke FYI

ErtugrulEge, 2025-09-03:
Please, double click to dimension  text and  just click ok. Shape will deform.

[offset_test.zip](https://github.com/user-attachments/files/22118715/offset_test.zip)

maxwxyz, 2025-09-03:
Confirmed

[image]

chennes, 2025-09-17:
I can reproduce this problem, but I have no idea what the appropriate solution is: there aren't any constraints being violated or anything. The geometry is just underconstrained and moves unexpectedly. This is probably something that either @PaddleStroke or @AjinkyaDahale would know much more about.

PaddleStroke, 2025-09-19:
The 'blowing up' appears to be due to some of the tangent constraints reversing : 

[image]

After some more test, it appears that this is due to the 'Auto-scale' on first dimension.

https://github.com/user-attachments/assets/64bde5db-f460-49f8-89a0-a5720a3994e5

closed 2025-09-19 by chennes