FreeCAD/FreeCAD #23578: PD Draggers: Cannot move the dragger when looking from the top view

maxwxyz, 2025-09-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I cannot drag on the draggers for Part Design features when looking normal to the profile of the feature:

[image]

### Full version info

```shell
OS: macOS 13.7.8
Architecture: x86_64
Version: 1.1.0dev.43241 (Git)
Build date: 2025/09/02 13:12:39
Build type: Unknown
Branch: main
Hash: 3e73005653be56fba18bd3d7328e60b500824211
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/72/1
Installed mods: 
  * freecad.gears 1.3.0
  * CfdOF 1.31.4
  * Defeaturing 1.2.6
  * Gridfinity 0.11.9
  * AddonManager 2025.7.17
  * Manipulator 1.6.0
  * fasteners 0.5.39
  * lattice2 1.0.0
  * Telemetry 1.0.2
  * CurvedShapes 1.0.13
  * Cables 0.2.1
  * FusedFilamentDesign 0.25.250
  * Curves 0.6.64
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-02:
@captain0xff FYI

maxwxyz, 2025-09-02:
From the bottom I can drag it until it glitches to the side. When orbiting it flips to the other side and then it breaks the angle:

https://github.com/user-attachments/assets/5314bbee-abb2-4802-8908-9df65a88dfb0

luzpaz, 2025-10-03:
@captain0xff will a fix for this make it into 1.1 ?

captain0xff, 2026-02-22:
But that gizmo rotates rather than translating. And you can't really do that from the top view. Also, the fact that the position is undefined while viewed from the top view doesn't help (Why? Because the arc in the gizmo faces the camera but it can't do that in the top view). So, in which direction do you really expect to pull it?

maxwxyz, 2026-02-22:
@captain0xff do you know whether it is possible for a unified solution as this affects also label movement and assembly movement in normal view:
- https://github.com/FreeCAD/FreeCAD/issues/13537