FreeCAD/FreeCAD #23599: PartDesign: Linear pattern seems to be broken

FEA-eng, 2025-09-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Unless I'm incorrectly using the new features, it seems that #22389 broke linear pattern.

Steps to reproduce:
1. Add a new Body.
2. Create Additive Box and select it.
3. Enable linear pattern.
4. There is an error message: `<Exception> No direction reference specified`
5. Set Direction 1 and Direction 2 to Base X-axis and Base Y-axis. Still the same errors keep appearing and you can't effectively add more occurrences.
6. Click Ok. A popup window with generic "Error" message appears:

[image]

Although the pattern is actually applied in the preview.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

courteousadorable2, 2025-09-03:
Download and test this.

maxwxyz, 2025-09-03:
@PaddleStroke FYI

Pendelf, 2025-09-03:
If you select "Select reference," the body disappears and the direction changes

https://github.com/user-attachments/assets/1a77029b-8e90-4a1f-b68f-e270214cb01f

PaddleStroke, 2025-09-03:
Thanks for the report and sorry for the regression. I'll check it out asap.

Conscientiia, 2025-09-08:
Hello, I also would like to add a minor bug that changing the dimensions doesn't update the preview overlay unless you interact with the model like zooming in or out or navigate the view. Also, it has an issue relating to "3D shading artifact" issue I suppose.

[image]

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Object center
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.814/1.25
Installed mods: 
  * addFC 1.9.2
  * Cables 0.2.1
  * Curves 0.6.70
  * fasteners 0.5.41
  * freecad.gears 1.3.0
  * Nodes 0.1.36
  * OpenTheme 2025.5.20
  * Quetzal 1.5.6
  * SearchBar 1.5.2.1
  * sheetmetal 0.7.51