FreeCAD/FreeCAD #23612: PartDesign: Pad/Pocket using TwoLengths and UpToShape broken

jffmichi, 2025-09-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

#21794 removed "TwoLengths" from `Pad::TypeEnums` and `Pocket::TypeEnums`. This means that all files that were created before this PR and use eitehr "TwoLengths" or "UpToShape" (came after "TwoLengths") are loaded incorrectly due to the enum value being interpreted differently. Here is a file that demonstrates the issue:

[partdesign_padpocket_issue.zip](https://github.com/user-attachments/files/22123900/partdesign_padpocket_issue.zip)

[image]

Left is the file as it was created in 1.0.2. Right is the file loaded in the current main. The Pad that uses "TwoLengths" gets wrongly treated as "UpToShape". The Pad001 that uses "UpToShape" gives an error.

I think this needs to be fixed as soon as possible because otherwise users who use the current main will save their files in a potentially corrupted way.

@PaddleStroke @kadet1090 

### Full version info

```shell
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.43259 (Git)
Build date: 2025/09/03 05:17:24
Build type: Unknown
Branch: main
Hash: ec8cff9e5fd6fb569a77ef70120701b184629849
Python 3.13.7, Qt 6.9.2, Coin 4.0.5, Vtk 9.5.1, boost 1_88, Eigen3 3.4.0, PySide 6.9.2
shiboken 6.9.2, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 8.0.0.rc2-5c814b0f0f
Locale: English/United States (en_US) [ OS: German/Germany (de_DE) ]
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/breeze
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7555/1.5
Installed mods: 
  * kicadStepUpMod 11.7.3
  * Gridfinity 0.11.9
  * sheetmetal 0.7.24
  * fcgear 1.0.0
  * fasteners 0.5.39
  * Assembly4 0.60.2
  * Curves 0.6.65
  * lattice2 1.0.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-09-03:
Thanks for the report and sorry for the regression. I'll check it out asap.

PaddleStroke, 2025-09-05:
The issue is that if we put in back in the enum, then we have to handle the UI so that it removes this item from the list. And more difficult is the property view. The user would still be able to set 'TwoLength' in there... How to handle that then?

kadet1090, 2025-09-05:
The code uses `?` as prefix to temporarily mark enum entries as invalid. Maybe we could just do `"?TwoLengths"` and then in the `getEnumVector` filter out these values? 

If we go with `Type1` and `Type2` then `Type` will still need to be different enum so we can transfer values without the shift. 

@hyarion maybe you have some thoughts? I don't see any good solution here...

closed 2025-09-10 by kadet1090

Rahix, 2025-09-19:
I think this is related: I just got a report that the `TwoLengths` change broke feature generation in my addon: https://github.com/Rahix/FusedFilamentDesign/issues/31

I guess the way forward from my side is to check freecad versions to decide what properties to set.

But I would suggest regardless that upstream FreeCAD should gracefully handle code setting `Type = TwoLengths` to not break existing addon/macro code.  Maybe auto-translate `Type = TwoLengths` into `SideType = Two sides` as is done on file-loading already, plus emitting a deprecation warning?