FreeCAD/FreeCAD #23624: SubShapeBinder Bind copy on change and suppressed features

fivethreeo, 2025-09-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a body with four features, each feature can be toggled from a property in the body. 

I create a subshapebinder in another body with “Bind copy on change" set to "Enabled".

 I drop the "template body" on the subshshapebinder and toggle all the features off. The last one I toggle off still shows. 

If I add one more feature to the "template body" that can’t be toggled and do the same thing,  all can be toggled off leaving the extra feature.

SubShapeBinder does not accept a empty body?

### Full version info

```shell
Rev 14555 Git
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

fivethreeo, 2025-09-04:
[BindCopyEnabled.zip](https://github.com/user-attachments/files/22138698/BindCopyEnabled.zip)

Updated issue with info.

maxwxyz, 2025-09-05:
The provided file is emtpy

fivethreeo, 2025-09-05:
[BindCopyEnabled.zip](https://github.com/user-attachments/files/22182992/BindCopyEnabled.zip)
/

fivethreeo, 2025-09-06:
[BindCopyEnabled.zip](https://github.com/user-attachments/files/22189158/BindCopyEnabled.zip)

There, now should be somehing there. Just drop TemplateBinder onto TemplatedBinder and toggle off all feature flags.

FEA-eng, 2025-11-06:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251104 (Git shallow)
Build date: 2025/11/04 19:07:44
Build type: Release
Branch: main
Hash: af014a1bb44264d749ba9522917f0d1ac76c4bbf
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

luzpaz, 2025-11-29:
Anybody able to offer a solution ?