FreeCAD/FreeCAD #23628: Sketcher: Offset tool crashes when committing value with Enter/Tab

longrackslabs, 2025-09-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce
	1.	Open a new sketch.
	2.	Select geometry and activate the Offset tool.
	3.	Type a numeric value in the inline editor.
	4.	Hit Enter or Tab.

Observed behavior
	•	FreeCAD immediately crashes.

Expected behavior
	•	Value should commit and the offset should apply without crashing.

Notes
	•	Clicking the canvas instead of pressing Enter/Tab does not crash.
	•	Might be related to #23058 (heap-use-after-free in EditableDatumLabel), but this repro path is different.

### Full version info

```shell
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 72/151.5/2
Installed mods: 
  * Gridfinity 0.11.9
  * manifest.json
  * fasteners 0.5.41
  * FusedFilamentDesign 0.25.250
  * Curves 0.6.70
```

### Subproject(s) affected?

None

### Anything else?

SketcherGui::StateMachine<SketcherGui::StateMachines::OneSeekEnd>::isLastState() const
SketcherGui::DrawSketchController<...>::finishControlsChanged()
SketcherGui::DrawSketchController<...>::tryViewValueChanged(int, double)
SketcherGui::DrawSketchController<...>::onViewValueChanged(int, double)
SketcherGui::DrawSketchController<...>::initNOnViewParameters(int)::'lambda'(double)
QtPrivate::FunctorCall<...>::call(...)
Gui::EditableDatumLabel::valueChanged(double)
Gui::EditableDatumLabel::startEdit(double, QObject*, bool)::$_0::operator()() const
...

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

chennes, 2025-09-08:
Running with ASAN on we get some more details:

```
__sanitizer::internal__exit(int) 0x00007ffc5ea8e90d
__sanitizer::Die() 0x00007ffc5ea90187
__asan::ScopedInErrorReport::~ScopedInErrorReport() 0x00007ffc5ead412e
__asan::ReportGenericError(unsigned long long,unsigned long long,unsigned long long,unsigned long long,bool,unsigned long long,unsigned int,bool) 0x00007ffc5ead5a8c
__asan_report_load8(unsigned long long) 0x00007ffc5ead8c6a
SketcherGui::DrawSketchController::finishControlsChanged() DrawSketchController.h:635
SketcherGui::DrawSketchController::tryViewValueChanged(int, double) DrawSketchController.h:437
SketcherGui::DrawSketchController::onViewValueChanged(int, double) DrawSketchController.h:358
`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1>::operator()(double) DrawSketchController.h:667
`QtPrivate::FunctorCall<std::integer_sequence<unsigned __int64,0>,QtPrivate::List<double>,void,`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1> >::call'::`2'::<lambda_1>::operator()() qobjectdefs_impl.h:116
QtPrivate::FunctorCallBase::call_internal<void,`QtPrivate::FunctorCall<std::integer_sequence<unsigned __int64,0>,QtPrivate::List<double>,void,`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1> >::call'::`2'::<lambda_1> >(void **,<lambda_1> &&) qobjectdefs_impl.h:65
QtPrivate::FunctorCall<std::integer_sequence<unsigned __int64,0>,QtPrivate::List<double>,void,`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1> >::call(<lambda_1> &,void **) qobjectdefs_impl.h:115
QtPrivate::FunctorCallable<`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1>,double>::call<QtPrivate::List<double>,void>(<lambda_1> &,void *,void **) qobjectdefs_impl.h:337
QtPrivate::QCallableObject<`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1>,QtPrivate::List<double>,void>::impl(int,QtPrivate::QSlotObjectBase *,QObject *,void **,bool *) qobjectdefs_impl.h:547
```
Here's the complete ASAN output:
```
G:\FreeCAD\FreeCAD\cmake-build-asan\bin\FreeCAD.exe
=================================================================
==25980==ERROR: AddressSanitizer: heap-use-after-free on address 0x11f822ec5c50 at pc 0x7ffc5585c2bf bp 0x00686b8f8320 sp 0x00686b8f8328
READ of size 8 at 0x11f822ec5c50 thread T0
    #0 0x7ffc5585c2be in SketcherGui::DrawSketchController<class SketcherGui::DrawSketchHandlerOffset, enum SketcherGui::StateMachines::OneSeekEnd, 0, class SketcherGui::OnViewParameters<1, 1>, enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::finishControlsChanged(void) G:\FreeCAD\FreeCAD\src\Mod\Sketcher\Gui\DrawSketchController.h:635
    #1 0x7ffc5589b95f in SketcherGui::DrawSketchController<class SketcherGui::DrawSketchHandlerOffset, enum SketcherGui::StateMachines::OneSeekEnd, 0, class SketcherGui::OnViewParameters<1, 1>, enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::tryViewValueChanged(int, double) G:\FreeCAD\FreeCAD\src\Mod\Sketcher\Gui\DrawSketchController.h:437
    #2 0x7ffc5588a6bc in SketcherGui::DrawSketchController<class SketcherGui::DrawSketchHandlerOffset, enum SketcherGui::StateMachines::OneSeekEnd, 0, class SketcherGui::OnViewParameters<1, 1>, enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::onViewValueChanged(int, double) G:\FreeCAD\FreeCAD\src\Mod\Sketcher\Gui\DrawSketchController.h:358
    #3 0x7ffc5582e5e6 in `protected: void __cdecl SketcherGui::DrawSketchController<class SketcherGui::DrawSketchHandlerOffset, enum SketcherGui::StateMachines::OneSeekEnd, 0, class SketcherGui::OnViewParameters<1, 1>, enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters(int)'::`4'::<lambda_1>::operator()(double) const G:\FreeCAD\FreeCAD\src\Mod\Sketcher\Gui\DrawSketchController.h:667
    #4 0x7ffc55829d66 in `public: static void __cdecl QtPrivate::FunctorCall<struct std::integer_sequence<unsigned __int64, 0>, struct QtPrivate::List<double>, void, class `protected: void __cdecl SketcherGui::DrawSketchController<class SketcherGui::DrawSketchHandlerOffset, enum SketcherGui::StateMachines::OneSeekEnd, 0, class SketcherGui::OnViewParameters<1, 1>, enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters(int)'::`4'::<lambda_1>>::call(class `protected: void __cdecl SketcherGui::DrawSketchController<class Sketcher [... cut ...]
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==25980==ABORTING
Exception: Exception 0x80000003 encountered at address 0x7ffc5ea8e90d

Process finished with exit code 0
```

chennes, 2025-09-08:
I did some basic debug stepping and found the control path that is resulting in the underlying handler being deallocated (though this is not being properly handled up the call stack, various stored pointers still refer to the now-dead memory:
```
SketcherGui::ViewProviderSketch::deactivateHandler() ViewProviderSketch.cpp:705
SketcherGui::ViewProviderSketch::purgeHandler() ViewProviderSketch.cpp:715
SketcherGui::DrawSketchDefaultHandler::handleContinuousMode() DrawSketchDefaultHandler.h:572
SketcherGui::DrawSketchDefaultHandler::finish() DrawSketchDefaultHandler.h:520
SketcherGui::DrawSketchDefaultHandler::onModeChanged() DrawSketchDefaultHandler.h:648
SketcherGui::DrawSketchControllableHandler::onModeChanged() DrawSketchControllableHandler.h:167
SketcherGui::StateMachine::setState(OneSeekEnd) DrawSketchDefaultHandler.h:142
SketcherGui::DrawSketchDefaultWidgetController::doChangeDrawSketchHandlerMode() DrawSketchHandlerOffset.h:1216
SketcherGui::DrawSketchController::finishControlsChanged() DrawSketchController.h:631
SketcherGui::DrawSketchController::tryViewValueChanged(int, double) DrawSketchController.h:437
SketcherGui::DrawSketchController::onViewValueChanged(int, double) DrawSketchController.h:358
`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1>::operator()(double) DrawSketchController.h:667
`QtPrivate::FunctorCall<std::integer_sequence<unsigned __int64,0>,QtPrivate::List<double>,void,`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1> >::call'::`2'::<lambda_1>::operator()() qobjectdefs_impl.h:116
QtPrivate::FunctorCallBase::call_internal<void,`QtPrivate::FunctorCall<std::integer_sequence<unsigned __int64,0>,QtPrivate::List<double>,void,`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1> >::call'::`2'::<lambda_1> >(void **,<lambda_1> &&) qobjectdefs_impl.h:65
QtPrivate::FunctorCall<std::integer_sequence<unsigned __int64,0>,QtPrivate::List<double>,void,`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1> >::call(<lambda_1> &,void **) qobjectdefs_impl.h:115
QtPrivate::FunctorCallable<`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1>,double>::call<QtPrivate::List<double>,void>(<lambda_1> &,void *,void **) qobjectdefs_impl.h:337
QtPrivate::QCallableObject<`SketcherGui::DrawSketchController<SketcherGui::DrawSketchHandlerOffset,enum SketcherGui::StateMachines::OneSeekEnd,0,SketcherGui::OnViewParameters<1,1>,enum SketcherGui::ConstructionMethods::OffsetConstructionMethod>::initNOnViewParameters'::`4'::<lambda_1>,QtPrivate::List<double>,void>::impl(int,QtPrivate::QSlotObjectBase *,QObject *,void **,bool *) qobjectdefs_impl.h:547
```

felixlungu, 2025-09-10:
Same happens on mac aarch64. Version 1.0 is not affected by this bug.