FreeCAD/FreeCAD #23651: PartDesign: (Sub)ShapeBinder can’t reference datum

pierreporte, 2025-09-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Create two bodies in a document. In one of them, create a datum (e.g. a plane). In the other body, try to create a ShapeBinder or a SubShapeBinder linked to that datum plane: it doesn’t work (there is no error message). Expected behavior: a (Sub)ShapeBinder is created.

[image]

Referencing datum and not just geometry is very useful when building complex models.

Partial workaround: make a (Sub)ShapeBinder from a surface that is coplanar to the desired plane, then create a datum plane based on the (Sub)ShapeBinder.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * sheetmetal-backup-before-zip-migration 0.7.22 (Disabled)
  * Curves-backup-before-zip-migration 0.6.63 (Disabled)
  * FusedFilamentDesign-backup-before-zip-migration 0.25.220 (Disabled)
  * fasteners 0.5.41
  * Curves 0.6.70
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.24
  * manifest.json
  * FusedFilamentDesign 0.25.250
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-09-05:
I guess it feasible yes

pierreporte, 2025-10-07:
@maxwxyz is this a regression?

luzpaz, 2025-11-29:
Is this a regression ?