FreeCAD/FreeCAD #23652: PartDesign: (Sub)ShapeBinders can’t be isolated

pierreporte, 2025-09-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

(Sub)ShapeBinders are reference geometry that belong to other bodies or solids, thus creating a permanent link between tho things. It can be desirable to break that link, especially to avoid cyclic dependencies, so that the (Sub)ShapeBinder becomes just a non-editable surface, curve, etc. There is no option do break that link to make them isolated from their parent.

The user may be willing to create (Sub)ShapeBinders that are automatically isolated by default. What I propose is:

1. Provide the user with a feature to isolate (Sub)ShapeBinders. For example an entry in the context menu.
2. Provide an option in the settings to either keep the link by default or break it immediatly.
3. For ShapeBinders, add a checkbox to override the default settings.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/117.596/1
Installed mods: 
  * sheetmetal-backup-before-zip-migration 0.7.22 (Disabled)
  * Curves-backup-before-zip-migration 0.6.63 (Disabled)
  * FusedFilamentDesign-backup-before-zip-migration 0.25.220 (Disabled)
  * fasteners 0.5.41
  * Curves 0.6.70
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.24
  * manifest.json
  * FusedFilamentDesign 0.25.250
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-09-04:
> There is no option do break that link to make them isolated from their parent.

Do you mean clearing the `Support` property? If so, it is indeed neither intuitive nor discoverable. But possible:

[image]

The `Support` property for SubShapebinders is read-only by default. By making it read/write, it can then be changed to point to another object or to clear it.

pierreporte, 2025-09-04:
@furgo16 Is rather the Bind Mode property that’s relevant. I missed it while reading the documentation and browsing the property view. It already has the feature, it’s the Detached mode. A bit easier to find but still not comfortable to use. However it only works for SubShapeBinders. ShapeBinders have far less properties, but according to #13044 there is no added value for them except a task panel.

pierreporte, 2025-09-05:
Only partially because there wouldn't be a easy access command for detaching nor the proposed settings. But it would be a fix for #13044.

furgo16, 2025-09-05:
I see what you mean. But in terms of easy access:

- Context menu:
  - Access via: 1 right click to show menu + left click to detach
  - Pros: quicker to access
  - Cons: the context menu keeps growing

- Task panel
  - Acess via: 1 double left click + left click to select detach option + 
  - Pros: consistent UX, allows for more properties to be added, implements the only added value from a ShapeBinder, opening the door for deprecation and duplicity removal
  - Cons: less quick to access

My thoughts were more along the lines of simplification/streamlining of UX and code. As well as killing two birds (well, one ShapeBinder) with one stone.