FreeCAD/FreeCAD #23663: UI/UX: Changing classic to dark theme gives "Requested non-existent style parameter" token error messages

ErtugrulEge, 2025-09-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everybody!

Maybe @furgo16 has similar or the same issue but I couldn't find the issue. 

Steps to reproduce:
1.Open FC with classic theme
2.Change theme to dark

https://github.com/user-attachments/assets/e93d9061-2c8d-4696-a97c-140541a624ed

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

MisterMakerNL, 2025-09-05:
These tokens were added in the latest pr. Not sure what happened. There was also a regression on the disabled checkmarks so ether the installation fails a bit or something went wrong with the pr.

JohnOCFII, 2025-10-04:
These messages also pop up on a fresh installation of FreeCAD.  Had to take a photo of the screen, as on default initial settings, the message disappears quickly, or if I attempt to move the mouse, etc.)

[image]

```
OS: macOS 15.7.1
Architecture: arm64
Version: 1.1.0dev.43830 (Git)
Build date: 2025/10/03 17:20:10
Build type: Unknown
Branch: main
Hash: eba045064768f334741c54bc3bb02dbf0c766a1c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
```

semhustej, 2025-11-05:
Tested with the latest weekly in safe mode. The issue is fixed. Thank you!

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251104 (Git shallow)
Build date: 2025/11/04 19:07:44
Build type: Release
Branch: main
Hash: af014a1bb44264d749ba9522917f0d1ac76c4bbf
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.3.post2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
  * UpdatedPythonPackages

```