FreeCAD/FreeCAD #23666: UI/UX: When changing dark theme to classic  doesn't affect the properties panel tab background color

ErtugrulEge, 2025-09-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to everyone!
Steps to reproduce:
1.Change classic them to dark
2.Change dark theme to classic
3.Draw any object and select you will see properties panel background is black
4.Restart FC everything turns to normal

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-08:
@kadet1090 @MisterMakerNL is this addressed with the recent PR?

ErtugrulEge, 2026-11-02:
Still relevant 

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 27.1.0dev.20261027 (Git shallow)
Build date: 2026/10/27 18:08:37
Build type: Release
Branch: (HEAD detached at 6135077d5)
Hash: 6135077d5c2d7c93850db34f23cb0e70f015b747
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 597.30/NVIDIA Corporation/Quadro RTX 4000/PCIe/SSE2
Installed mods:
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.82
```