FreeCAD/FreeCAD #23697: GUI: Report View font cannot be changed from Courier

Spire42, 2025-09-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Under Windows 11, it is impossible to change the _Report View_ font (which is also used in the Python console and the macro editor) from the default Courier (or possibly Courier New).

I am able to select a different font in the _Preferences_ dialog box, and my selection is correctly used in the preview there, but in the app itself the font used is always Courier.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 144/161.962/1
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I first observed this problem in the [2025-07-29 weekly build](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.07.29). The last time it worked correctly for me was the [2025-07-26 weekly build](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.07.26).

I suspect commit fcd1b92, which ostensibly fixed #12199 under macOS, but may have also introduced this regression under Windows (and possibly elsewhere).

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-09-06:
See https://github.com/FreeCAD/FreeCAD/issues/22882#issuecomment-3258533936

Spire42, 2025-09-06:
Interesting. Issue #22882 describes a completely different symptom but does appear to have been caused by the same commit, fcd1b92.

I have two questions:

1. Does my specific issue affect mac OS?
2. If it does, then doesn't your proposed hotfix #23699, which wraps the new Mac-specific code in `#ifdef` guards, fail to fix my issue on mac OS?

Perhaps instead of a hotfix, it would be best for now to completely revert the changes made in fcd1b92, as suggested by @leematos [here](https://github.com/FreeCAD/FreeCAD/issues/22882#issuecomment-3148869858).

Syres916, 2025-09-06:
@Spire42 if the commit is **fully** reverted then Windows and Linux users will be happy but all those Mac users who have just got used to their system not producing warning(s) about `Courier not available` will start raising Issues here and/or on the forum. My hotfix is intended to try to keep _everyone_ happy. Your issue is easily provable by the next Weekly if merged by Monday evening which I'm extremely confident will be fixed by the hotfix.

Spire42, 2025-09-06:
I have no doubt that my issue would be resolved by your hotfix under Windows and Linux. What I am wondering is whether my issue currently exists under mac OS — because if it does, your hotfix would prevent it from being fixed there.

I have no way of testing the current weekly build under mac OS, so I can't say.