FreeCAD/FreeCAD #23702: Flatpak beta updates

madbobmcjim, 2025-09-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Is there any chance we could get more frequent updates to the Flatpak beta? I'd love to be able to use it to test new features :-)

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.40267 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 04d2bcb)
Hash: 04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical/physical DPI: 96/108.877
Installed mods: 
  * Gridfinity 0.11.9
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-11-15:
@adrianinsaval can you transfer to the flatpak repo please ?

luzpaz, 2025-12-02:
Closing ticket. Please follow the progress at 
* https://github.com/flathub/org.freecad.FreeCAD/issues/34
* https://github.com/flathub/org.freecad.FreeCAD/pull/40