FreeCAD/FreeCAD #23721: Assembly: add ability to view joints attached to a selected part

eblanshey, 2025-09-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It is a common task to identify how a part is attached to the assembly. Currently you have to go to the joint list, click on each joint, and see if it includes the part you are interested in. Or you could rename each joint to include the part names, but that can get unwieldy with a lot of joints.

A better solution that other CAD software have is a panel that opens when selecting a part in Assembly workbench. The panel displays all the joints touching the selected part. Combined with https://github.com/FreeCAD/FreeCAD/issues/18527 it would make Joint review very simple.

### Full version info

```shell
v1.1dev
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-08:
@PaddleStroke isn't that the same as you are working on?

PaddleStroke, 2025-11-21:
Done.
But it needs those 2 PRs to merge first before I can open the PR : 
https://github.com/FreeCAD/FreeCAD/pull/25513
https://github.com/FreeCAD/FreeCAD/pull/24623

https://github.com/user-attachments/assets/c857d31c-d066-4835-b3e9-7e6b559d7779