FreeCAD/FreeCAD #23723: CAM: PySide2 and new UI Tool Controller editing

Dimitris75, 2025-09-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm trying to test the new UI for editing tool controller parameters from within operation and I'm getting the following error. (related to  #23180 ?)

`
File "/freecad-source/build/Mod/CAM/Path/Tool/Gui/Controller.py", line 256, in selectInComboBox
    with QtCore.QSignalBlocker(combo):
<class 'TypeError'>: 'PySide2.QtCore.QSignalBlocker' object does not support the context manager protocol
`

https://github.com/FreeCAD/FreeCAD/blob/2aac5f6b3361ad9f93c027238569ade88447969e/src/Mod/CAM/Path/Tool/Gui/Controller.py#L230-L264

By removing the With statement in line 256 I'm receiving the same error from line 233, again related to  PySide2.QtCore.QSignalBlocker' and the context manager protocol.

Maybe there is an issue with Qt5 and context manager?

### Full version info

```shell
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.43285 (Git)
Build date: 2025/09/05 19:07:02
Build type: Release
Branch: main
Hash: 140c9febc49be4f861a590e4765f58e23299af8f
Python 3.13.5, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7727/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Connor9220, 2025-09-10:
@davidgilkaufman Can you look at this?  This is because it's using pyside2 / qt5.

davidgilkaufman, 2025-09-14:
This PR should fix it #23916