FreeCAD/FreeCAD #23725: Rotate BIM Section Plane doesn't change Cut View

psommerfeld, 2025-09-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As in attached project:
1. Create BIM Section Plane on a BIM Wall
2. Select View > Cut View
3. Notice everything about the plane is cut
4. Double-click the Section Plane
5. In the Task Panel, rotate around X axis 180 degrees so that the corner arrows point upward 
6. Turn off Cut View and re-enable to refresh
7. Note that the same part of the Wall is still cut, therefore not honoring the Section Plane direction

[image]

[image]

[techraw dims-bim-section.FCStd.zip](https://github.com/user-attachments/files/22197335/techraw.dims-bim-section.FCStd.zip)

### Full version info

```shell
OS: macOS 15.6.1
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paullee0, 2025-09-07:
Seems there had been discussion that Section Plane does work well when the direction is pointing upward - some problem to generate reflected ceiling plan to my memory, probably related problem.

semhustej, 2025-09-08:
I encountered this issue too. It is a shame it is not possible to rotate 180 degrees, not sure what the reason is

There is work being done by Yorik on Reflected Ceiling Plan tool: https://github.com/FreeCAD/FreeCAD/issues/5738

Roy-043, 2025-09-08:
This code segment is responsible for the current behavior:
https://github.com/FreeCAD/FreeCAD/blob/45629f75994e1afb08f94ddb0a5aba0b54cafc12/src/Mod/BIM/ArchSectionPlane.py#L867-L870
Replacing these 4 lines with the code below will fix the issue, but there may be unexpected side-effects.
```python
        p = Part.makePlane(l,h,Vector(-l/2,-h/2,0))
```

semhustej, 2025-09-08:
> This code segment is responsible for the current behavior:
> 
> [FreeCAD/src/Mod/BIM/ArchSectionPlane.py](https://github.com/FreeCAD/FreeCAD/blob/45629f75994e1afb08f94ddb0a5aba0b54cafc12/src/Mod/BIM/ArchSectionPlane.py#L867-L870)
> 
> Lines 867 to 870 in [45629f7](/FreeCAD/FreeCAD/commit/45629f75994e1afb08f94ddb0a5aba0b54cafc12)
>  p = Part.makePlane(l,h,Vector(l/2,-h/2,0),Vector(0,0,-1)) 
>  # make sure the normal direction is pointing outwards, you never know what OCC will decide... 
>  if p.normalAt(0,0).getAngle(obj.Placement.Rotation.multVec(FreeCAD.Vector(0,0,1))) > 1: 
>      p.reverse() 
> 
> Replacing these 4 lines with the code below will fix the issue, but there may be unexpected side-effects.
> 
>         p = Part.makePlane(l,h,Vector(-l/2,-h/2,0))

I think @yorikvanhavre would know, why the 180 degree rotation of the section was disabled.