FreeCAD/FreeCAD #23733: Addon Manager crashes FreeCAD

Jnorichards, 2025-09-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The first time I'm using Addon Manager having moved up from 1.0.1 to 1.0.2. **T**ools, **A**ddon Manager opens a window and the 'Loading addon information' progress bar reaches 44% and stops. A further popup dialog opens inviting me to enter SSH credentials for github.com, which in my case I have not got.
Closing the SSH dialog leaves Addon Manager frozen: the Close button is inoperative. Closing the Addon Manager window with the GUI contraol gives an OS message that Addon Manager is not responding, and FreeCAD is terminated.
The console (from which freecad was started) has message which says "Stopped".  fg command restarts with result:
`/tmp/.mount_freecaPgoLhn/AppRun: line 31: 186446 Killed                  "${MAIN}" "$@"` 
and the credentials dialog only is re-displayed.

### Full version info

```shell
OS: KDE neon User Edition (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7e
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:

Help 1.0.3
Assembly4 0.50.15
freecad.gears 1.2.0
fasteners 0.5.28
ExplodedAssembly
LCInterlocking
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Jnorichards, 2025-09-07:
Self-reply - I find that if I just click 'OK' in the credentials dialog, a further prompt is issued, but having given a null password to both, Addon Manager will load. So, still an issue but not as dire as I thougt.
Jonathan

xtemp09, 2025-09-08:
I don't what exactly is the problem, but it is possible that your IP is DDoSing github.com, so it asks for credentials and drops the connection (or the entire subnet) if they are absent.

Of course, _Addon Manager_ should not crash in this case.

luzpaz, 2025-11-23:
@chennes move this to addon manager repo ?

chennes, 2025-11-23:
Only if it still occurs in 1.1 or 1.2 -- it's still flagged as "needs test on dev version" mostly because the Addon Manager no longer directly talks to GitHub, so it's unlikely to still occur, at least in the reported form.

closed 2025-11-23 by maxwxyz