FreeCAD/FreeCAD #23738: Sketcher - Move/Array Transform less than unity (1.0) input in inch mode results in integer

DougSmithShop, 2025-09-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

See attached file.  I am trying to sketch a bellows.  I am using FreeCAD 1.1 weekly build 42975 and 2025.08.03 on Linux laptops.  In inch mode, 5 decimal places I try to enter 9 copies spaced 0.08" at 90 degrees.  As soon as i enter a decimal point, a zero and an 8, it changes the value to 8 inches or whatever the first non-zero digit.  It will only allow a decimal value equal to whatever I have stretched  the copies.  When I try to change the value, it ignores the decimal and used the first non-zero digit as an integer value.  This is not true in ISO (metric) mode where it works correctly.  It also works correctly in inch mode in FreeCAD 1.0.2

### Full version info

```shell
OS: fossapup64 9.5 ()
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.357/1
Installed mods: 
  * fasteners 0.5.38
  * FreeCAD-Documentation-html 1.0.0-alpha
  * ThreadProfile 1.94.0
also
OS: fossapup64 9.5 ()
Architecture: x86_64
Version: 1.1.0dev.42975 (Git) AppImage
Build date: 2025/08/18 17:31:10
Build type: Release
Branch: main
Hash: bafd9f75b50633cf533e13e55b00116be53adc2b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.357/1
Installed mods: 
  * fasteners 0.5.38
  * FreeCAD-Documentation-html 1.0.0-alpha
  * ThreadProfile 1.94.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

[send.FCStd.zip](https://github.com/user-attachments/files/22200074/send.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-08:
@PaddleStroke FYI

PaddleStroke, 2025-09-08:
@tetektoza

tetektoza, 2025-10-20:
@maxwxyz I think this one was resolved recently. Are you able to recheck?

closed 2025-10-21 by maxwxyz