FreeCAD/FreeCAD #23760: Creating or adding to a VarSet through the double click dialog produces a defective console transcript

pyro9, 2025-09-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating or adding to a VarSet, the commands echoed to the Python console are incomplete and too verbose.
For example, create a VarSet and add the property Test of type App::PeopertyInteger setting it's value to 50. I might expect the console to read something like:
>>> ### Begin command Std_VarSet
>>> App.activeDocument().addObject('App::VarSet','VarSet')
>>> App.ActiveDocument.getObject('VarSet').ViewObject.doubleClicked()
>>> ### End command Std_VarSet
>>> FreeCAD.getDocument('Unnamed3').getObject('VarSet').addProperty('App::PropertyInteger', 'Test', 'Base', '')
>>> FreeCAD.getDocument('Unnamed3').getObject('VarSet').Test=5

 #===========
Instead, I get:
>>> ### Begin command Std_VarSet
>>> App.activeDocument().addObject('App::VarSet','VarSet')
>>> App.ActiveDocument.getObject('VarSet').ViewObject.doubleClicked()
>>> ### End command Std_VarSet
>>> FreeCAD.getDocument('Unnamed3').getObject('VarSet').Te = ""
>>> 
>>> FreeCAD.getDocument('Unnamed3').getObject('VarSet').Tes = ""
>>> 
>>> FreeCAD.getDocument('Unnamed3').getObject('VarSet').Test = ""
>>> 
>>> FreeCAD.getDocument('Unnamed3').getObject('VarSet').Test = 0
>>> 
>>> FreeCAD.getDocument('Unnamed3').getObject('VarSet').Test = 5
>>> 
>>> FreeCAD.getDocument('Unnamed3').getObject('VarSet').Test = 50
>>> 

Note that where it goes from ="" to =0 is where I set the type to PropertyInteger

This affects 1.0.0 and recent dev versions at least.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/26 20:37:46
Build type: Release
Branch: main
Hash: a5b3a2caac8feb34070415b637b5fbc2a3043a48
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 93/95.1083/1
Installed mods: 
  * fasteners 0.5.38
  * Curves 0.6.63
  * 3D_Printing_Tools
  * Silk 1.0.0
  * Assembly4 0.50.18
  * A2plus 0.4.68
  * CfdOF 1.21.6
  * fcgear.backup1677967119.4973533 1.0.0 (Disabled)
  * fcgear 1.0.0
  * sheetmetal 0.2.61
  * Render 2023.8.18
  * CurvedShapes 1.0.13
  * lattice2 1.0.0
  * FourAxis
  * ThreadProfile.backup1749851552.5109692 1.90.0 (Disabled)
  * ThreadProfile 1.96.0
  * FusedFilamentDesign 0.25.250
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-09-09:
cc @pieterhijma

pieterhijma, 2025-09-09:
Yes, thanks for reporting. That's in issue indeed.