FreeCAD/FreeCAD #23763: Core: Deleting a Body/Part crashes FC.

leoheck, 2025-09-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I just built the latest main. The process went fine.
Deleting a Body/Part crashes FC.

How to reproduce:
1. Create a new file.
2. Create a Body/Part (it did not crash when using a Group)
3. Delete it

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.43405 (Git)
Build date: 2025/09/09 04:20:15
Build type: Unknown
Branch: main
Hash: b400cdcec7275cfacc537efe502d1da8baf50a73
Python 3.13.3, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * sheetmetal 0.7.24
  * DynamicData 2.74.0
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * osh-autodoc-workbench 0.2.3
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.3
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Pendelf, 2025-09-09:
It looks like everything is okay

https://github.com/user-attachments/assets/441ca44d-326b-40bb-8fed-38081c2e2957

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC 1.9.1
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * fasteners 0.5.41
  * PieMenu 1.12.0
  * sheetmetal 0.7.24
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.2

Syres916, 2025-09-09:
@Pendelf your build doesn't capture all of Monday's merge though, I'm just self-compiling up-to-date and will report back.

@leoheck ref `Build type: Unknown` you really should have `-DCMAKE_BUILD_TYPE=Release` in your _CMake_ command (or the `Debug` equivalent) otherwise your not actually replicating the user builds.

Syres916, 2025-09-09:
OK, trace is:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3e540) [0x7fe90783e540]
#1  /usr/lib/libc.so.6(+0x18f1d0) [0x7fe90798f1d0]
#2  0x7fe90d39c3cc in Gui::ViewProviderPlane::getRole[abi:cxx11]() const from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x2c
#3  0x7fe90d39c648 in Gui::ViewProviderPlane::updatePlaneSize() from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x198
#4  0x7fe90d51979b in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x2b
#5  0x7fe90d0f8527 in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x347
#6  0x7fe90d51c74f in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x30f
#7  0x7fe90d0f8527 in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x347
#8  0x7fe90d51b73d in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x28d
#9  0x7fe90d526175 in Gui::SelectionSingleton::slotDeletedObject(App::DocumentObject const&) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x545
#10  0x7fe90bd6ef27 in boost::signals2::detail::signal_impl<void (App::DocumentObject const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::DocumentObject const&)>, boost::function<void (boost::signals2::connection const&, App::DocumentObject const&)>, boost::signals2::mutex>::operator()(App::DocumentObject const&) from /home/username/freecad-daily-build/lib/libFreeCADApp.so+0x297
#11  0x7fe90bfcc180 in App::Application::slotDeletedObject(App::DocumentObject const&) from /home/username/freecad-daily-build/lib/libFreeCADApp.so+0x10
#12  0x7fe90bd6ef27 in boost::signals2::detail::signal_impl<void (App::DocumentObject const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::DocumentObject const&)>, boost::function<void (boost::signals2::connection const&, App::DocumentObject const&)>, boost::signals2::mutex>::operator()(App::DocumentObject const&) from /home/username/freecad-daily-build/lib/libFreeCADApp.so+0x297
#13  0x7fe90bd2f334 in App::Document::_removeObject(App::DocumentObject*, Base::Flags<App::RemoveObjectOption>) from /home/username/freecad-daily-build/lib/libFreeCADApp.so+0x194
#14  0x7fe90bdf4d32 in App::DocumentPy::removeObject(_object*) from /home/username/freecad-daily-build/lib/libFreeCADApp.so+0x72
#15  0x7fe90bdf4ff3 in App::DocumentPy::staticCallback_removeObject(_object*, _object*) from /home/username/freecad-daily-build/lib/libFreeCADApp.so+0x23
#16  /usr/lib/libpython3.13.so.1.0(+0x196213) [0x7fe90a596213]
#17  /usr/lib/libpython3.13.so.1.0(_PyObject_MakeTpCall+0x2bb) [0x7fe90a5620cb]
#18  /usr/lib/libpython3.13.so.1.0(_PyEval_EvalFrameDefault+0x40da) [0x7fe90a57697a]
#19  /usr/lib/libpython3.13.so.1.0(PyEval_EvalCode+0xa9) [0x7fe90a64e2c9]
#20  /usr/lib/libpython3.13.so.1.0(+0x28c88c) [0x7fe90a68c88c]
#21  /usr/lib/libpython3.13.so.1.0(+0x28985d) [0x7fe90a68985d]
#22  /usr/lib/libpython3.13.so.1.0(+0x284cde) [0x7fe90a684cde]
#23  0x7fe90b7e5966 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/username/freecad-daily-build/lib/libFreeCADBase.so+0x66
#24  0x7fe90ce0d167 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x87
#25  0x7fe90ce2e002 in StdCmdDelete::activated(int) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x11f2
#26  0x7fe90ce13089 in Gui::Command::_invoke(int, bool) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x259
#27  0x7fe90ce13536 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x176
#28  /usr/lib/libQt6Core.so.6(+0x1d466f) [0x7fe9081d466f]
#29  0x7fe908e07800 in QAction::activate(QAction::ActionEvent) from /usr/lib/libQt6Gui.so.6+0x210
#30  0x7fe90ce929ff in Gui::ShortcutManager::onTimer() from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x12f
#31  0x7fe90ce9392e in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0x57e
#32  0x7fe90ce94be6 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /home/username/freecad-daily-build/lib/libFreeCADGui.so+0xb6
#33  0x7fe9081679f8 in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0xd8
#34  0x7fe909301df4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0xb4
[... cut ...]
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.13.7, Qt 6.9.2, Coin 4.0.5, Vtk 9.5.1, boost 1_88, Eigen3 3.4.0, PySide 6.9.2
shiboken 6.9.2, SMESH 7.7.1.0, xerces-c 3.3.0, IfcOpenShell 0.8.3.post2, OCC 7.9.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.9416/1

```

kadet1090, 2025-09-09:
Interesting, I'll checkout seems like pretty random crash but probably something is captured in the lambda.

kadet1090, 2025-09-09:
> The current weekly is also crashing a lot. I am working with Sketcher, and sometimes when closing the sketch makes it crash. It might be related.
> 
> ```
> FreeCAD 1.1.0, Libs: 1.1.0devR14555 (Git shallow)
> (C) 2001-2025 FreeCAD contributors
> FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
> 
> Object can only be in a single GeoFeatureGroup
> <App> OriginGroupExtension.cpp(209): Reset origin in ID_Simplified#Body001
> Chamfer008: BRep_API: command not done
> Chamfer008: BRep_API: command not done
> Program received signal SIGSEGV, Segmentation fault.
> #0  /lib/x86_64-linux-gnu/libc.so.6(+0x458d0) [0x70c854c458d0]
> #1  0x70c80110ef54 in Geom2dAPI_ProjectPointOnCurve::Init(gp_Pnt2d const&, opencascade::handle<Geom2d_Curve> const&) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/libTKGeomAlgo.so.7.8+0x14
> #2  0x70c80110efcd in Geom2dAPI_ProjectPointOnCurve::Geom2dAPI_ProjectPointOnCurve(gp_Pnt2d const&, opencascade::handle<Geom2d_Curve> const&) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/libTKGeomAlgo.so.7.8+0x3d
> #3  0x70c803372bd2 in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/libTKFillet.so.7.8+0x2ae2
> #4  0x70c80337bf15 in ChFi3d_Builder::PerformOneCorner(int, bool) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/libTKFillet.so.7.8+0x2da5
> #5  0x70c803319f3b in ChFi3d_Builder::PerformFilletOnVertex(int) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/libTKFillet.so.7.8+0x58b
> #6  0x70c80331af73 in ChFi3d_Builder::Compute() from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/libTKFillet.so.7.8+0xf93
> #7  0x70c80349abb6 in BRepFilletAPI_MakeChamfer::Build(Message_ProgressRange const&) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/libTKFillet.so.7.8+0x36
> #8  0x70c801669a4d in BRepBuilderAPI_MakeShape::Shape() from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/libTKTopAlgo.so.7.8+0x8d
> #9  0x70c803bd9725 in Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, char const*) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/Part.so+0x75
> #10  0x70c803bed4cb in Part::TopoShape::makeElementChamfer(Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> > const&, Part::ChamferType, double, double, char const*, Part::Flip) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/Part.so+0x4bb
> #11  0x70c76cbccd09 in PartDesign::Chamfer::execute() from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/_PartDesign.so+0x419
> #12  0x70c858f95f6b in App::DocumentObject::recompute() from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x6b
> #13  0x70c80391f60a in Part::Feature::recompute() from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/Part.so+0xa
> #14  0x70c76cb141e0 in PartDesign::Feature::recompute() from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/lib/_PartDesign.so+0x160
> #15  0x70c858f4990c in App::Document::_recomputeFeature(App::DocumentObject*) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x2bc
> #16  0x70c858f47aa8 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0xba8
> #17  0x70c85900d196 in App::DocumentPy::recompute(_object*) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x346
> #18  0x70c859005758 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libFreeCADApp.so+0x48
> #19  /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x1faeb8) [0x70c8577faeb8]
> #20  /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2b4) [0x70c8577d9c04]
> #21  /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6e7) [0x70c8577e53c7]
> #22  /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x29ef1d) [0x70c85789ef1d]
> #23  /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x70c85789e914]
> #24  /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2bced3) [0x70c8578bced3]
> #25  /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8aea) [0x70c8578b8aea]
> #26  /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x70c8578acfe9]
> #27  0x70c85897f0c1 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/lheck/Applications/FreeCAD/weekly/2025.09.03/squashfs-root/usr/bin/../lib/libFreeCADBase.so+0x71
[... cut ...]
> Installed mods: 
>   * Assembly4.1 0.60.2.1
>   * Cables 0.2.1
>   * CurvesWB 0.6.64
>   * FreeCAD-render 2024.12.15
>   * sheetmetal 0.7.24
>   * DynamicData 2.74.0
>   * InventorLoader 1.5.1
>   * diff-tools 0.1.0
>   * ose-piping
>   * OpenTheme 2025.5.20
>   * fasteners 0.5.41
>   * osh-autodoc-workbench 0.2.3
>   * freecad-povray-render
>   * OpticsWorkbench 1.2.2
>   * lattice2 1.0.0
>   * kicadStepUpMod 11.7.3
> ```

looking at the stack it seems to be a different issue, more likely in the OCC.

leoheck, 2025-09-09:
Ok, I will move to a different issue. I crashed it again. I will try to create a procedure to replicate the crash.

kadet1090, 2025-09-10:
Fixed by: https://github.com/FreeCAD/FreeCAD/pull/23791

closed 2025-09-10 by kadet1090