FreeCAD/FreeCAD #23766: Creating a Hole in Part Design from a point in a sketch results in "Cannot make face from profile"

K4LCIFER, 2025-09-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If you try to create a hole from a point, it will fail with the error "Cannot make face from profile". The full error is as follows:
```
{'sclassname': 'N4Base14CADKernelErrorE', 'sErrMsg': 'Cannot make face from profile', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Cannot make face from profile', 'btranslatable': False, 'breported': False}
```

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 256fc7e)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Anything else?

The ability to create a hole from a point was added here: https://github.com/FreeCAD/FreeCAD/issues/6884

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Pendelf, 2025-09-09:
I cannot confirm

https://github.com/user-attachments/assets/e2b670fa-16c2-4ae1-852f-fba98c0f0211

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * addFC 1.9.1
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * fasteners 0.5.41
  * PieMenu 1.12.0
  * sheetmetal 0.7.24
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.2

pierreporte, 2025-09-09:
The feature is not available in 1.0.2. It will be in 1.1.

K4LCIFER, 2025-09-09:
> The feature is not available in 1.0.2. It will be in 1.1.

Noted. I'll close this then.

Pendelf, 2025-09-09:
@K4LCIFER 

Partly yes, but a similar interface can also be set up in version 1.0. I recommend downloading the latest development version 1.1.

Here’s a video on interface settings:

https://youtu.be/aWG7sNQgZ7g