FreeCAD/FreeCAD #23772: Part Scale: Keyboard navigation is broken.

leoheck, 2025-09-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

All the widgets should have keyboard navigation reviewed.

This is one example of it.
The whole thing is impossible to navigate in a sane way.
A special problem is when scaling Non-Uniformly.
I cannot navigate to the next item easily.

From the X factor, the selection disappears when hitting tab,  and one more click leads me to the Z factor.

[image]

### Full version info

```shell
1.1-dev (probably since forever)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-11-14:
The tab order is not defined in the .ui file, in which case the default is to use the order in which the widgets are defined in the file. Since they're not defined sequentially in the way they are rendered, then the tab order is all over the place. Here's what it looks like in Qt Designer's Tab Order Editor.

[image]

furgo16, 2025-11-14:
Indeed, feel free to. Happy to tackle them if I can.