FreeCAD/FreeCAD #23800: PD Thickness: Pulling the dragger overrides the expression on the Thickness field.

leoheck, 2025-09-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Create a Body
1. Create a Sketch
1. Pad it
1. Select the top Face
1. Apply the Thickness
1. Enter = and add a custom equation (e.g., = 2 mm)
1. Pull the Dragger

_Result: The Thickness value changes directly in the Thickness field._

Possible fixes:

1. Pulling the dragger should reset the field it controls. Note: This may only be acceptable if a confirmation pop-up appears; otherwise, the user could lose the expression.
1. If there is an expression, the dragger should appear locked with a symbol or color.
1. Alternatively, the dragger could be hidden when an expression is present.

Note, I am reporting this on this PD Feature, but it may happen on other places too.

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * sheetmetal 0.7.24
  * DynamicData 2.74.0
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * osh-autodoc-workbench 0.2.3
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

captain0xff, 2025-09-10:
@leoheck I am able to reproduce this. The draggers should be hidden automatically if the corresponding property has a expression bound to it.

[image]

leoheck, 2025-09-10:
@captain0xff it indeed does what you said, except that if you close it and open it again, the issue appears. I will update the steps to reproduce it in the first message.

When creating, if I add the Expression, the dragger gets hidden.

[image]

Once you close the feature and open it again, the dragger appears, and I can override the expression.

[image]

After that, the field will show the value of the dragger... but the applied value will always be the expression. This is very confusing for the user. But a single fix, to check if there is an expression when going to edit the feature.