FreeCAD/FreeCAD #23801: PD Thickness: Mode, Join Type, Intersection could have tooltips

leoheck, 2025-09-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

PD Thickness: Mode, Join Type, Intersection could have tooltips explaining what they do, including the choices we have.

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * sheetmetal 0.7.24
  * DynamicData 2.74.0
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.41
  * osh-autodoc-workbench 0.2.3
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.3
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-09-10:
Sometimes I have to use Intersection, but I have no clue why and what it does to work.

FEA-eng, 2025-09-10:
Mode has no effect, only Skin works (#6885).

FEA-eng, 2025-09-10:
> [@hyarion](https://github.com/hyarion) didn't you remove this? Maybe not in the thickness tool?

That PR only affected the Offset 3D tool and was reverted because of a bug.

drewhamiltonasdf, 2026-09-11:
This PR can be closed for a variety of reasons?

maxwxyz, 2026-11-07:
This is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/588eecc300667986ec8d608220b27310ddda6f32

The three controls have tooltips explaining their choices and the limitation of the intersection option.