FreeCAD/FreeCAD #23808: New sketch: Access validation

maxwxyz, 2025-09-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is a recent regression:

1. Open Part Design
2. Click on new sketch
3. Access violation

An empty body is created but no sketch.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-10:
@PaddleStroke @kadet1090 any idea what might have caused this?

Syres916, 2025-09-10:
Certainly fixed using:

```
OS: Debian GNU/Linux 13 (trixie) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.43410 (Git)
Build date: 2025/09/10 08:43:48
Build type: Release
Branch: main
Hash: 2073ccf8e2e47ee65f8344521a7ec1714dd1c2b8
Python 3.13.5, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 6.8.2.1
shiboken 6.8.2.1, SMESH 7.7.1.0, xerces-c 3.2.4, IfcOpenShell 0.8.3.post2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/96.0757/1
Installed mods: 
  * FreeCAD-themes 2025.6.4
  * AddonManager 2025.8.16
```