FreeCAD/FreeCAD #23811: Core: the Clarify Selection feature could list Shells, Solids, Compounds and CompSolids too

FEA-eng, 2025-09-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, the Clarify Selection tool, based on the Realthunder's Pick Geometry, lists the following object types:
- Object
- Face
- Edge
- Vertex
- Other

It would be great if it also listed Shells, Solids, Compounds and CompSolids like Pick Geometry in the Link Branch:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-09-18:
@FEA-eng bodies also do not show up, bot the last feature of them.

hans-eat-ic, 2026-01-16:
Could "Clarify Selection" be implemented into Sketcher as in this video?
https://www.youtube.com/watch?v=vUZezMhwwow

FEA-eng, 2026-01-16:
> Could "Clarify Selection" be implemented into Sketcher as in this video? https://www.youtube.com/watch?v=vUZezMhwwow

From what I know, @tetektoza is working on that.