FreeCAD/FreeCAD #23827: SIGSEGV on new sketch

Beep6581, 2025-09-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

1. Launch FreeCAD
2. Close Start tab
3. Create new document ctrl+n
4. In Part Design, create new sketch. SIGSEGV

```
#0  0x00007ffff4d18ce0 in App::PropertyLink::getValue() const () from /home/morgan/programs/freecad/lib/libFreeCADApp.so
#1  0x00007ffff4c4f0b1 in App::OriginGroupExtension::getOrigin() const () from /home/morgan/programs/freecad/lib/libFreeCADApp.so
#2  0x00007fff9eda7e92 in PartDesignGui::SketchWorkflow::tryCreateSketch() () from /home/morgan/programs/freecad/lib/PartDesignGui.so
#3  0x00007fff9eda9e5c in PartDesignGui::SketchWorkflow::createSketch() () from /home/morgan/programs/freecad/lib/PartDesignGui.so
#4  0x00007fff9ed7577d in CmdPartDesignNewSketch::activated(int) () from /home/morgan/programs/freecad/lib/PartDesignGui.so
#5  0x00007ffff5c16c9d in Gui::Command::_invoke(int, bool) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#6  0x00007ffff5c16fb6 in Gui::Command::invoke(int, Gui::Command::TriggerSource) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#7  0x00007ffff5c170fa in Gui::GroupCommand::activated(int) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#8  0x00007ffff5c16b09 in Gui::Command::_invoke(int, bool) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#9  0x00007ffff5c16fb6 in Gui::Command::invoke(int, Gui::Command::TriggerSource) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#10 0x00007ffff5beeaf3 in Gui::ActionGroup::onActivated() () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#11 0x00007ffff0dd37ef in ?? () from /usr/lib/libQt6Core.so.6
#12 0x00007ffff1a073c0 in QAction::activate(QAction::ActionEvent) () from /usr/lib/libQt6Gui.so.6
#13 0x00007ffff20423ad in ?? () from /usr/lib/libQt6Widgets.so.6
#14 0x00007ffff20425c1 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /usr/lib/libQt6Widgets.so.6
#15 0x00007ffff2185721 in QToolButton::mouseReleaseEvent(QMouseEvent*) () from /usr/lib/libQt6Widgets.so.6
#16 0x00007ffff1f5b5c7 in QWidget::event(QEvent*) () from /usr/lib/libQt6Widgets.so.6
#17 0x00007ffff1f01c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt6Widgets.so.6
#18 0x00007ffff1f05c39 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt6Widgets.so.6
#19 0x00007ffff5b8e678 in Gui::GUIApplication::notify(QObject*, QEvent*) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#20 0x00007ffff0d68118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt6Core.so.6
#21 0x00007ffff1ef7358 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /usr/lib/libQt6Widgets.so.6
#22 0x00007ffff1f76e13 in ?? () from /usr/lib/libQt6Widgets.so.6
#23 0x00007ffff1f780d8 in ?? () from /usr/lib/libQt6Widgets.so.6
#24 0x00007ffff1f01c70 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt6Widgets.so.6
#25 0x00007ffff5b8e678 in Gui::GUIApplication::notify(QObject*, QEvent*) () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#26 0x00007ffff0d68118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt6Core.so.6
#27 0x00007ffff15a1217 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/lib/libQt6Gui.so.6
#28 0x00007ffff16248f4 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt6Gui.so.6
#29 0x00007fffeb4f90c7 in ?? () from /usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6
#30 0x00007fffed108880 in g_main_dispatch (context=0x7fffe4000f70) at ../glib/glib/gmain.c:3398
#31 0x00007fffed109cd7 in g_main_context_dispatch_unlocked (context=0x7fffe4000f70) at ../glib/glib/gmain.c:4249
#32 g_main_context_iterate_unlocked (context=context@entry=0x7fffe4000f70, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4314
#33 0x00007fffed109ee5 in g_main_context_iteration (context=0x7fffe4000f70, may_block=1) at ../glib/glib/gmain.c:4379
#34 0x00007ffff103c5e2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt6Core.so.6
#35 0x00007ffff0d744b6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt6Core.so.6
#36 0x00007ffff0d6c7c1 in QCoreApplication::exec() () from /usr/lib/libQt6Core.so.6
#37 0x00007ffff5a99fe0 in Gui::Application::runApplication() () from /home/morgan/programs/freecad/lib/libFreeCADGui.so
#38 0x000055555555b7f2 in main ()
```

Debug info coming soon.

### Full version info

```shell
689740d3d3
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Beep6581, 2025-09-10:
```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.43415 (Git)
Build date: 2025/09/10 18:07:42
Build type: Unknown
Branch: main
Hash: 689740d3d3f6465ff2df20657713ff07b2ca049b
Python 3.13.7, Qt 6.9.1, Coin 4.0.3, Vtk 9.5.1, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1.5
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.41
  * AddonManager 2025.8.27
  * CurvedShapes 1.0.13
  * woodworking 1.1.0.dev.1.1.1.2025.09.10
  * Curves 0.6.70
```

```
#0  0x00007ffff32d2a9a in App::PropertyLink::getValue (this=0xb58) at /home/morgan/programs/code-freecad/src/App/PropertyLinks.cpp:759
#1  0x00007ffff3204817 in App::OriginGroupExtension::getOrigin (this=0x958) at /home/morgan/programs/code-freecad/src/App/OriginGroupExtension.cpp:58
#2  0x00007fff9aba8194 in (anonymous namespace)::PlaneFinder::tryFindBasePlanes (this=0x7fffffffa450)
    at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:472
#3  0x00007fff9aba7880 in (anonymous namespace)::PlaneFinder::findBasePlanes (this=0x7fffffffa450)
    at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:399
#4  0x00007fff9aba8ec2 in (anonymous namespace)::SketchRequestSelection::findAndSelectPlane (this=0x7fffffffa440)
    at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:623
#5  0x00007fff9aba84f9 in (anonymous namespace)::SketchRequestSelection::tryFindSupport (this=0x7fffffffa440)
    at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:540
#6  0x00007fff9aba83dc in (anonymous namespace)::SketchRequestSelection::findSupport (this=0x7fffffffa440)
    at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:516
#7  0x00007fff9abaa59e in PartDesignGui::SketchWorkflow::tryCreateSketch (this=0x7fffffffa9f0)
    at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:814
#8  0x00007fff9aba9f53 in PartDesignGui::SketchWorkflow::createSketch (this=0x7fffffffa9f0)
    at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:772
#9  0x00007fff9ab60e74 in CmdPartDesignNewSketch::activated (this=0x555557fcc930, iMsg=0)
    at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/Command.cpp:516
#10 0x00007ffff51e8978 in Gui::Command::_invoke (this=0x555557fcc930, id=0, disablelog=false) at /home/morgan/programs/code-freecad/src/Gui/Command.cpp:431
#11 0x00007ffff51e87dd in Gui::Command::invoke (this=0x555557fcc930, i=0, trigger=Gui::Command::TriggerNone)
    at /home/morgan/programs/code-freecad/src/Gui/Command.cpp:406
#12 0x00007ffff51ec18c in Gui::GroupCommand::activated (this=0x555557fccb50, iMsg=0) at /home/morgan/programs/code-freecad/src/Gui/Command.cpp:1045
#13 0x00007ffff51e8ab4 in Gui::Command::_invoke (this=0x555557fccb50, id=0, disablelog=true) at /home/morgan/programs/code-freecad/src/Gui/Command.cpp:442
#14 0x00007ffff51e87dd in Gui::Command::invoke (this=0x555557fccb50, i=0, trigger=Gui::Command::TriggerAction)
    at /home/morgan/programs/code-freecad/src/Gui/Command.cpp:406
#15 0x00007ffff51c636d in Gui::ActionGroup::onActivated (this=0x55555829b2f0) at /home/morgan/programs/code-freecad/src/Gui/Action.cpp:583
```

```
#0  0x00007ffff32d2a9a in App::PropertyLink::getValue (this=0xb58) at /home/morgan/programs/code-freecad/src/App/PropertyLinks.cpp:759
No locals.
#1  0x00007ffff3204817 in App::OriginGroupExtension::getOrigin (this=0x958) at /home/morgan/programs/code-freecad/src/App/OriginGroupExtension.cpp:58
        originObj = 0x7fffffffa000
#2  0x00007fff9aba8194 in (anonymous namespace)::PlaneFinder::tryFindBasePlanes (this=0x7fffffffa450) at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:472
        origin = 0x1007ffff4f85ae5
#3  0x00007fff9aba7880 in (anonymous namespace)::PlaneFinder::findBasePlanes (this=0x7fffffffa450) at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:399
No locals.
#4  0x00007fff9aba8ec2 in (anonymous namespace)::SketchRequestSelection::findAndSelectPlane (this=0x7fffffffa440) at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:623
        appdocument = 0x55555e3692e0
        planes = std::vector of length 17592155805713, capacity 11728130002610 = {<error reading variable planes (Cannot access memory at address 0x0)>
        status = std::vector of length -11728123352066, capacity -60473868 = {4294943360, 32767, 4079036147, 32767, 1479439760, 21845, 4294943440, 32767, 4294943504, 32767, 4059756919, 32767, 2599254552, 32767, 4061580675, 32767, 1434332424, 21845, 1483940976, 21845, 1434395280, 21845, 65535, 
          PartDesignGui::TaskFeaturePick::noWire, 4294943472, 32767, 4111634063, 32767, PartDesignGui::TaskFeaturePick::validFeature, PartDesignGui::TaskFeaturePick::validFeature, 1441993552, 21845, PartDesignGui::TaskFeaturePick::noWire, PartDesignGui::TaskFeaturePick::noWire, 1441993552, 21845, 4294943504, 
          32767, 1441993552, 21845, PartDesignGui::TaskFeaturePick::invalidShape, PartDesignGui::TaskFeaturePick::otherBody, 1441993552, 21845, 4294943536, 32767, 1431843118, 21845, 1919251285, 1918988320, 1441993552, 21845, 4294943568, 32767, [... cut ...]
#8  0x00007fff9aba9f53 in PartDesignGui::SketchWorkflow::createSketch (this=0x7fffffffa9f0) at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/SketchWorkflow.cpp:772
No locals.
#9  0x00007fff9ab60e74 in CmdPartDesignNewSketch::activated (this=0x5555564725d0, iMsg=0) at /home/morgan/programs/code-freecad/src/Mod/PartDesign/Gui/Command.cpp:516
        creator = {guidocument = 0x55555e406fa0, appdocument = 0x55555e3692e0, activeBody = 0x0}
#10 0x00007ffff51e8978 in Gui::Command::_invoke (this=0x5555564725d0, id=0, disablelog=false) at /home/morgan/programs/code-freecad/src/Gui/Command.cpp:431
        manager = 0x555555ea4ec0
```

depthoffocus, 2025-09-10:
Duplicate of #23814 I think?