FreeCAD/FreeCAD #23828: BIM: SectionPlane cuts grid

furgo16, 2025-09-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When the SectionPlane's `CutView` property is set to `True` and the section's plane intersects the working plane where the grid is placed on, the grid is cut as well.

[image]
Example: grid on the XY plane, section plane at 90°, grid is cut

Expected: grid should not be cut

### Full version info

```shell
1.1.0-dev
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-09-19:
Std_ToggleClipPlane has the same issue. Everything in the scengraph is cut, Including the axis cross and the tracker previews of Draft command.

furgo16, 2025-09-25:
I do agree with your points. Fixing this is beyond my abilities though, hence the proposal of a pragmatic workaround. I still think is a valid option to consider, but if someone with more Coin3D expertise jumps in to work on a proper fix, that'd indeed be ideal.